stlehmann/pyads #15: Reading Boolean inside an FB which is instantiated in an Array

seehma, 2016-10-03:
Hi,

First, thanks for this python library. It makes it very easy to access PLC variables.
I have one problem accessing a Boolean inside an FB which is instantiated in an Array.
I have the following scenario:

- I have an FB calls IoOutput with a output_ AT%Q* : BOOL instantiated inside
- This IoOutput FB is instantiated as an Array of three outputs (output_[0], output_[1], output_[2])
- And this Array is instantiated in another FB and so on to finally hosted in the MAIN Prg

Now i want to access the Boolean inside the IoOutput FB but that does not work
I tried the following call: redLight = pyads.read_by_name(adr, 'MAIN.ampel1_.output_.output_[0].output_', pyads.PLCTYPE_BOOL)

Did you try such an example => Accessing Variables in FBs which are instantiated as Array?

Hope you understand my problem, maybe with the pic attached...

Thanks,

Matthias
[image]
[image]

seehma, 2016-10-03:
OMG.

Beginners Mistake => The error Message was correct, i had one output_ too much in my Handle Name.

Sorry for that Issue, you can close it!

closed 2016-10-03 by seehma

MrLeeh, 2016-10-03:
Alright. Glad that things worked out :)