stlehmann/pyads #17: <AmsAddress 0.0.0.0.1.1:30000>

eSkiSo, 2017-01-13:
Hello, when i try to use **pyads.get_local_address()**, it always returns **<AmsAddress 0.0.0.0.1.1:30000>**, my issue is on the **0.0.0.0** instead of my ip, how can i get around this?
I don't want to have it as **0.0.0.0** because i need to add the ip to the **PLC**.

Thank you.

dabrowne, 2017-01-15:
Are you running on Linux? I've encountered this too, it seems to be a bit of a nuance with how the AMS Router works (part of the c++ library provided by Beckhoff). You have to add a route before it registers the IP address of the local machine. Try doing it in this order:

```python
pyads.open_port()
adr = pyads.AmsAddr('5.33.160.54.1.1', 851)
remote_ip = '192.168.0.100'
pyads.add_route(adr, remote_ip)

# This should now return the correct AMS address
pyads.get_local_address()
```

eSkiSo, 2017-01-17:
Yes, this works now thank you.
One questing, how can i ignore the "Connecting to.." outputs, i only want it to output what i print.
Is there a way to ignore this **LOG_INFO**?
`
2017-01-17T18:21:40+0000 Info: Connecting to [IP]

[returned value that i will output myself]

2017-01-17T18:21:40+0000 Info: connection closed by remote
`

dabrowne, 2017-01-17:
Those messages are being written to stderr by the logging module in the ADS lib. There should be a way to suppress this somehow. I'll have a look and get back to you.

closed 2017-04-06 by MrLeeh