stlehmann/pyads #19: Add device notification methods

pjaneck, 2017-07-11:
Hey,
I don't know if you're already working on this but it would be nice to have the methods "AdsSyncAddDeviceNotificationReq" & "AdsSyncDelDeviceNotificationReq" to use callbacks on variable changes.
I will try to implement them myself.
Regards,
Peter

MrLeeh, 2017-07-12:
Hello Peter,
you are very welcome to add the notification methods to the library.
Greetings Stefan

pjaneck, 2017-07-12:
Hey Stefan,
see the attached patch. It works on windows and my ubuntu virtual machine.
I didn't try it on python3, only python2.
Let me know what you think.
Regards,
Peter

[adding_notify_callbacks.patch.zip](https://github.com/MrLeeh/pyads/files/1142485/adding_notify_callbacks.patch.zip)

MrLeeh, 2017-07-14:
Hey Peter
thanks for the PR. I will have a look at it in time.
Regards

MrLeeh, 2017-07-15:
Hey Peter,

I looked a bit through your code and I like it very much. I remember wanting to implement the notification callback functionality myself but skipped it because I didn't know how to implement it at that time. 

However we will need some sort of testing for this. I think it should be implemented in the integration test module. Could you think of implementing the necessary tests yourself? Otherwise I will have a closer look at it. 

I also like the idea of the class approach. I think we should definetely implement this as well. However I would prefer a classname "ADSConnection" here as the device doesn't necessary need to be a plc.

Greetings
Stefan

closed 2017-07-18 by MrLeeh