stlehmann/pyads #26: Add examples for device notification

MrLeeh, 2017-07-24:


kirlek, 2017-08-31:
This would be nice since I am unable to access _data_ from pyads.structs.SAdsNotificationHeader

MrLeeh, 2017-08-31:
I agree, the handling of device notifications is a bit awkward. I want to take care of it in the near future.

pjaneck, 2017-09-07:
Let me help you:
```python
def callbackBool(self, adr, notification, user):
        contents = notification.contents
        var = map(bool, bytearray(contents.data)[0:contents.cbSampleSize])[0]
```
```python
def callbackInt(self, adr, notification, user):
        contents = notification.contents
        var = map(int, bytearray(contents.data)[0:contents.cbSampleSize])[0]
```
```python
def callbackString(self, adr, notification, user):
        dest = (c_ubyte * contents.cbSampleSize)()
        memmove(addressof(dest), addressof(contents.data), contents.cbSampleSize)
        # I had some NULL bytes in my string...
        var = str(bytearray(dest)).split('\x00')[0]
```
Regards,
Peter

kirlek, 2017-09-08:
Thank you!

For accessing a float I used:

`
import struct
def callbackFloat(adr, notification, user):
    contents = notification.contents
    var = struct.unpack('f', bytearray(contents.data)[0:contents.cbSampleSize])[0]
`

kirlek, 2017-09-08:
Now I just need to access the timestamp also.

According to [Beckhoff infosys](https://infosys.beckhoff.com/english.php?content=../content/1033/tcsample_labview/html/tcsample_labview_overview.htm&id=): "The time stamp is transferred in the FILETIME format."

I am unsure how to use the integer from _nTimeStamp_ since it is not in FILETIME format, or is it?

pjaneck, 2017-09-11:
Have a look at the C++ [examples](https://infosys.beckhoff.com/content/1033/tcsample_vc/html/tcadsdll_api_cpp_sample08.htm).
Since the timestamp is a 64bit integer I would assume they use an unix timestamp.

kirlek, 2017-09-11:
Beckhoff uses Windows FILETIME for their timestamps which differs from unix timestamps. 

> 64-bit FILETIME structure, which represents the number of 100-nanosecond intervals since January 1, 1601 UTC

However I am not sure how I should handle the python integer that is available in __nTimestamp__

For one event I get _6643331907373236323_ and the next time I run I get _15529404494432436324_

None of these are a Windows Filetime...

I wonder if _nTimeStamp_ should be of another type?
```
class SAdsNotificationHeader(Structure):
    _pack_ = 1
    _fields_ = [("nTimeStamp", c_uint64),   <-- wrong datatype?
                ("hNotification", c_uint32),
                ("cbSampleSize", c_uint32),
                ("data", POINTER(c_ubyte))]
```

As we can see in the example linked by __pjaneck__ in C++ the 64-bit value is converted into a TIMESTAMP.
```
  LargeInteger.QuadPart = pNotification->nTimeStamp;
  FileTime.dwLowDateTime = (DWORD)LargeInteger.LowPart;
  FileTime.dwHighDateTime = (DWORD)LargeInteger.HighPart;
```
Should pyads also do this? I tried using __ctypes.wintypes.FILETIME__ but had no success since I am unsure how to implement it. Just changing the type in __SAdsNotificationHeader__ still just returns an integer that is not a FILETIME.

MrLeeh, 2017-09-18:
@pjaneck Thank you for the Examples you provided here. I will include them in the documentation.

MrLeeh, 2017-09-19:
@kirlek In the [Beckhoff documentation ](https://infosys.beckhoff.de/content/1033/tcadsdll2/html/tcadsdll_strucadsnotificationheader.htm?id=7318830137739770883) the Timestamp is declared as a 64 Bit value. So this should be OK. I agree with yout that pyads should return a datetime.datetime value. I'm currently working on a decorator to handle the transformation.

MrLeeh, 2017-09-19:
@kirlek. Now I know what you mean. The timestamp doesn't make any sense at all. Tried a few ways to figure out how it is calculated. No success so far :(

MrLeeh, 2017-09-19:
This issue now gets handled in #33.

closed 2017-09-19 by MrLeeh