stlehmann/pyads #29: Linux: pyads uses wrong IP adress

carstenschroeder, 2017-07-27:
Hello!

First I would like to say thank you for providing pyads as open source! It fits great for my home automation project at home.

I found the following issue:
If the AMS addr. is not derived from the IP address of the remote maschine pyads seems to use a wrong IP address to connect .

I use the newest version of pyads from git from today.

Example:
Remote maschine:
AMS addr.: 5.2.102.179.1.1
IP addr: 192.168.6.21

The following calls lead to an error:

>>> import pyads
>>> pyads.open_port()
30000
>>> adr = pyads.AmsAddr('5.2.102.179.1.1', pyads.PORT_SPS1)
>>> pyads.add_route(adr, '192.168.6.21')
2017-07-27T09:04:36+0000 Info: Connecting to **192.168.6.21**           _<< correct_

>>> pyads.get_local_address()
<AmsAddress 192.168.6.27.1.1:30000>

>>> plc = pyads.Connection('5.2.102.179.1.1', pyads.PORT_SPS1)
>>> plc.open()
2017-07-27T09:04:53+0000 Info: Connecting to **5.2.102.179**               _<< false_
2017-07-27T09:07:04+0000 Error: Connect TCP socket failed with: 110
2017-07-27T09:07:04+0000 Info: connection closed by remote
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyads/ads.py", line 309, in open
    adsAddRoute(self._adr.netIdStruct(), self.ip_address)
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads_ex.py", line 75, in wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads_ex.py", line 99, in adsAddRoute
    raise ADSError(error_code)
pyads.pyads.ADSError: ADSError: Internal error (1)

carstenschroeder, 2017-07-27:
Ok, I found a workaround, the IP is an optional parameter:

plc = pyads.Connection('5.2.102.179.1.1', pyads.PORT_SPS1, '192.168.6.21')
plc.open()

now it works.

But shouldn't it take the IP from the routing definition as default?

MrLeeh, 2017-07-27:
Hello Carsten,
at the moment if you don't provide an IP address on initialization of the Connection class the ip address is simply derived from the AMS address by skipping the last two numbers. If your IP differs you need to pass it as a parameter. There is o need to add a route manually anymore. A route will be added automatically on `Connection.open()` and removed on `Connection.close()`.

carstenschroeder, 2017-07-27:
Ok, thank you for clarification!

closed 2017-07-27 by carstenschroeder