stlehmann/pyads #30: Linux: KeyError: 2127924964 on plc.close()

carstenschroeder, 2017-07-27:
On linux the following error occures:

>>> import pyads
>>> pyads.open_port()
30000
>>> plc = pyads.Connection('5.2.102.179.1.1', pyads.PORT_SPS1, '192.168.6.21')
>>> plc.open()
2017-07-27T10:31:42+0000 Info: Connecting to 192.168.6.21
>>> plc.read_by_name('.gWSM_Scene_ID_Buero', pyads.PLCTYPE_BYTE)
2
>>> plc.close()
2017-07-27T10:32:02+0000 Info: connection closed by remote
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyads/ads.py", line 322, in close
    adsDelRoute(self._adr.netIdStruct())
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads_ex.py", line 75, in wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads_ex.py", line 118, in adsDelRoute
    raise ADSError(error_code)
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads.py", line 34, in __init__
    self.msg = "{} ({})".format(ERROR_CODES[self.err_code], self.err_code)
KeyError: 2127924964

I use the newest version of pyads from git from today on raspbian.

MrLeeh, 2017-07-27:
I didn't experience that, yet. So I was not able to try it on a Raspberry Pi, yet. Can you try to skip the `pyads.open_port()` statement?

    import pyads
    plc = pyads.Connection('5.2.102.179.1.1', pyads.PORT_SPS1, '192.168.6.21')
    with plc:
        plc.read_by_name('.gWSM_Scene_ID_Buero', pyads.PLCTYPE_BYTE)

carstenschroeder, 2017-07-27:
The connection works also w/o the pyads.open_port() statement. But the error keeps the same.

>>> plc.close()
2017-07-27T11:19:02+0000 Info: connection closed by remote
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/pyads/ads.py", line 322, in close
    adsDelRoute(self._adr.netIdStruct())
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads_ex.py", line 75, in wrapper
    return fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads_ex.py", line 118, in adsDelRoute
    raise ADSError(error_code)
  File "/usr/local/lib/python2.7/dist-packages/pyads/pyads.py", line 34, in __init__
    self.msg = "{} ({})".format(ERROR_CODES[self.err_code], self.err_code)
KeyError: 2128137956

carstenschroeder, 2017-07-27:
I think I found the root cause:

The function AdsDelRoute does not return a return code

ADS/AdsLib/AdsLib.cpp:
...
void AdsDelRoute(const AmsNetId ams)
{
    router.DelRoute(ams);
}
...

So evaluate the return code on python side make no sense:
@router_function
def adsDelRoute(net_id):
    """
    :summary:  Remove existing route from the AMS Router.
    :param pyads.structs.SAmsNetId net_id: net id associated with the routing
        entry which is to be removed from the router.
    """

    delete_route = _adsDLL.AdsDelRoute
    delete_route.restype = ctypes.c_long

    error_code = delete_route(net_id)

    if error_code:
        raise ADSError(error_code)

MrLeeh, 2017-07-28:
You are right. That is a mistake that sneaked in. The `DelRoute()` function does not provide a return value.

MrLeeh, 2017-07-28:
OK, I merged your PR after also deleting the return type declaration. Can you have a look if it is working now and provide a short feedback.

carstenschroeder, 2017-07-28:
>>> import pyads
>>> plc = pyads.Connection('5.2.102.179.1.1', pyads.PORT_SPS1, '192.168.6.21')
>>> plc.open()
2017-07-28T06:16:02+0000 Info: Connecting to 192.168.6.21
>>> plc.read_by_name('.gWSM_Scene_ID_Buero', pyads.PLCTYPE_BYTE)
2
>>> plc.close()
2017-07-28T06:16:14+0000 Info: connection closed by remote

Works! Thanks a lot!

closed 2017-07-28 by carstenschroeder

MrLeeh, 2017-07-29:
Thanks for your contribution.