stlehmann/pyads #36: Install through pip fails with ELF CLASS error

lexe, 2017-09-28:
Hi,

This is just a minor bug. If i install your library through pip on a raspberry pi, the installation fails with an ELF CLASS error. (x86 vs x64) This happened since the adslib.so has been included as a shared library. I found a workaround be overwriting the adslib.so of an older version (2.1.0).

Regards

MrLeeh, 2017-09-28:
@lexe Thanks for reporting. I discovered the same bug when running on Mac OS X. Your workaround is OK as the adslib.so hasn't been changed between the versions. Another workaround would be to compile the adslib.so manually and insert it in the pyads package. What Python version are you working with?

lexe, 2017-09-28:
Im currently on 3.4.2.

regards

MrLeeh, 2017-10-03:
Actually this is bug is due to the usage of wheel files. Wheel files include the compiled version of the library which is OK if you' re on Linux as I precompiled the whole thing on Linux. However somehow they can not differentiate between Linux, Mac or Linux on Raspberry. So for now I will stop uploading wheel files and only provide the source distribution on PyPi. This way it is guaranteed that the ADS library is getting compiled on your os if you install it via pip install.

closed 2017-10-03 by MrLeeh