stlehmann/pyads #37: Different SAdsNotificationHeader for Windows and Linux ADS version

MrLeeh, 2017-09-29:
The arrangement of the fields in the struct SAdsNotificationHeader seems to differ between Windows ADS and Linux ADS. 

When using the adslib.so the struct is arranged this way:

    class SAdsNotificationHeader(Structure):
        _pack_ = 1
        _fields_ = [("hNotification", c_uint32),
                    ("nTimeStamp", c_uint64),
                    ("cbSampleSize", c_uint32),
                    ("data", POINTER(c_ubyte))]

On Windows however the parsing would show unfitting values for both nTimeStamp and hNotification. In fact both fields are swapped in the Windows version:

    class SAdsNotificationHeader(Structure):
        _pack_ = 1
        _fields_ = [("nTimeStamp", c_uint64),
                    ("hNotification", c_uint32)
                    ("cbSampleSize", c_uint32),
                    ("data", POINTER(c_ubyte))]

This issue has been already addressed on #33. @dabrowne Maybe it would be best to just swap the two fields in the adslib? What do you thing?

MrLeeh, 2017-09-29:
I made a pull request on https://github.com/dabrowne/ADS that swaps the fields. I tested it with pyads on Linux and on Windows. Looks everything fine now.

closed 2017-09-29 by MrLeeh

dabrowne, 2017-09-30:
Thanks Stefan, looks like a good solution. I've merged your PR.