stlehmann/pyads #47: In Linux when Connection Lost how to fix it amsroute

BurakAvci1912, 2018-02-23:
Hello, I am writing python3.5 with using library on raspberrypi.This Service read tags from plc periodicly.
But When connection lost between raspberry pi and beckhoff plc, again not fixed connection I am solving this issue using deleterouting addrouting.Is there any Correct solution ?
Main.py :
`from Clients.Client import BeckhoffClient
from time import sleep

plc = BeckhoffClient('5.22.252.149.1.1', 801, '192.168.1.24')
while True:
    try:
        if not plc.is_connected():
            success = plc.connect_plc()
            print('Connection Successfully')
    except Exception as ex:
        print(ex)
    sleep(10)
`
Client.py :
`import pyads

class BeckhoffClient:
    def __init__(self, amsNetId, port, targetIp):
        self.AmsNetID = amsNetId
        self.Port = port
        self.TargetIp = targetIp
        self.Connection = pyads.Connection(self.AmsNetID, self.Port, self.TargetIp)

    def connect_plc(self):
        try:
            adr = pyads.AmsAddr(self.AmsNetID, pyads.PORT_SPS1)
            pyads.delete_route(adr)
            pyads.add_route(adr, self.TargetIp)
            self.Connection.open()
            if self.Connection.is_open:
                self.Connection.read_device_info()

                return True
            else:
                return False
        except Exception as ex:
            self.Connection.close()
            raise ex

    def is_connected(self):
        if self.Connection.is_open:
            try:
                self.Connection.read_device_info()
                return True
            except Exception as ex:
                return False
        else:
            return False`

stlehmann, 2018-02-27:
I think that would be the way to go. If you want to go more pythonic you can make use of the Connection class which handles adding and deleting of a route automatically.

BurakAvci1912, 2018-02-27:
thank you for replies Stlehman. Give me your advice Connection Class details or examples ? :)

stlehmann, 2018-02-27:
Just have a look in the documentation or the Readme file. Usage is as easy as:

    import pyads
    plc = pyads.Connection('127.0.0.1.1.1', pyads.PORT_SPS1)
    plc.open()
    plc.close()

closed 2018-03-01 by stlehmann