stlehmann/pyads #50: Couldn't communicate.

TamakiAutomation, 2018-04-30:
Hello, I'm trying to communicate CX2020 with PC.

But I couldn't understand how to find the proper setting for getting global variables.

Network setting is follwings.

>  Name | AmsNetId | IP | Port
> Desktop       | 192.168.0.15.1.1 | 169.254.170.72
> EAP3            | 5.53.192.206.4.1 | 169.254.193.18 | 27908
> EAP2            | 5.53.192.206.3.1 | 169.254.37.249 | 27907
> EatherCat     | 5.53.192.206.2.1 |  27905
> Target NetId | 5.53.192.206.1.1 |  
> Local NetId  | 192.168.0.15.1.1

Pyads code is just simple.
`plc = pyads.Connection(5.53.192.206.3.1,27907,169.254.37.249)`
`plc.open()`
ADSError:timeout elapsed (1861)

I tried combination of AmsNetId and port like 48898,851.

How should I config network setting?

stlehmann, 2018-04-30:
Are you on Windows? If you are on a PC with Windows you need to have TwinCAT installed. And you will need to add the routes manually within your TwinCAT software for the connection to work.
Have a look here: https://infosys.beckhoff.com/english.php?content=../content/1033/tcsystemmanager/basics/TcSysMgr_AddRouteDialog.htm&id=

stlehmann, 2018-05-01:
If you are on Linux and you use the Connection class the route is added automatically. As you already added the route info on your remote machine one reason it doesn't work can be a wrong AMS-NetID - IP address combination on either side.

If I look at your IP-addresses I see that you local address (192.168.0.*) and your target (169.254.*.*) probably live in different subnets. So they are not able to see each other. Make sure the target also gets an IP address in the 192.168.0 subnet then it should work.

TamakiAutomation, 2018-05-01:
local address(192.168.0) is devloping PC with twincat 3.
Linux PC port is 169.254.193.20(netmask 255.255.0.0).
I can login CX2020 from Linux via Device manager tool, so it should be connected.

So I'm thinking also ams-net id, ip address, port.