stlehmann/pyads #52: Set own AmsNetID

SmilingWorlock, 2018-05-15:
Hello,

 I am trying to use pyads from within a docker container - this works fine, but since the AmsNetID is depentent on the IP Address, and the "internal" IP Address of a container is not necesseraly fixed, the AmsNetID changes a lot, resulting in routing problems on the remote device. Is there a way to set the AmsNetID indepently of the IP Address?

Best regards

stlehmann, 2018-05-15:
There is no function in the Beckhoff ADS DLL that let's you choose your AmsNetID.

dabrowne, 2018-05-15:
@stlehmann This functionality was added to upstream for the Linux lib here: https://github.com/Beckhoff/ADS/commit/c1bfedd25988468ac0f46e9cf0fff4d4be3ffa52

If you merge this commit (also present in my ADS fork) then we'd be able to provide this at least for Linux clients.

stlehmann, 2018-05-16:
@dabrowne I didn't know of this one. Thanks. I will merge the commit and add this function to pyads.

stlehmann, 2018-05-16:
@SmilingWorlock I added a new `set_local_address` function in the Release 2.2.13. Feel free to give it a go.

closed 2018-05-16 by stlehmann