stlehmann/pyads #53: Target port not found   ADS Server not started (6)

f-scotto, 2018-06-26:
Hi All, 
I am trying to control some variables on the PLC through pyads. Unfortunately I keep on receiving always the same error message. 'target port not found   ADS Server not started (6)'. Any ideas on how to get around this issue? Thanks a lot in advance. 

`import pyads

plc = pyads.Connection('10.2.9.142.1.1', 851)
plc.read_device_info()`

stlehmann, 2018-06-26:
Are you on Linux ord Windows?

f-scotto, 2018-06-26:
windows 10

stlehmann, 2018-06-26:
Did you install Twincat on your PC and setup the routes to your target properly?

f-scotto, 2018-06-27:
Yes, The route is setup correctly and the twinCAT is running on my machine correctly. I have uninstalled and installed the whole system but whenever i call the pyads.Connection('AMSNetId', 851) I get the same error message. (target port not found   ADS Server not started (6)). I cannot figure why!?

f-scotto, 2018-06-27:
If I run the TcAdsTest I get this error message (see pic). 
[image]

stlehmann, 2018-06-28:
This clearly looks like you've got an issue with the setup of your TwinCAT system. Unfortunately I can  not help you with this as it is not related to pyads. Maybe it helps to check if your Beckhoff ADS library is installed correctly on your system.

closed 2018-06-28 by stlehmann

Summerxiaxia623, 2022-04-13:
I got the same problem. Have you solved that?

ishahak, 2023-01-23:
**Solution that worked for me:** 
In TwinCAT, project explorer, right-click your project name under "PLC" branch, and click _Change ADS Port_. it should be 851 normally. if it is not, update your Connection initialization accordingly.

DuartePinguim, 2025-01-20:
Like @ishahak mention you need to go to your TwinCAT and see in what port PLC is working 

[image]

[image]

Other thing you can use the constants of pyads

[image]