stlehmann/pyads #55: Callback Function not working for Windows platform

ajaxjjj20, 2018-08-15:
When trying to add a device notification on a Windows computer, I am getting:
```
Traceback (most recent call last):
  File "ADSReporting.py", line 61, in <module>
    plc.add_device_notification('MAIN.i', pyads.NotificationAttrib(2), callback)

  File "C:\Users\ajacques\AppData\Local\Continuum\anaconda3\lib\site-packages\py
ads\ads.py", line 623, in add_device_notification
    user_handle)
  File "C:\Users\ajacques\AppData\Local\Continuum\anaconda3\lib\site-packages\py
ads\pyads_ex.py", line 585, in adsSyncAddDeviceNotificationReqEx
    raise TypeError("Callback function type can't be None")
TypeError: Callback function type can't be None
```

This error is coming from pyads_ex.py:
```    
if LNOTEFUNC is None:
        raise TypeError("Callback function type can't be None")
```

But it appears that LNOTEFUNC is only populated if the platform is Linux:
```
LNOTEFUNC = None

# load dynamic ADS library
if platform_is_windows():
    _adsDLL = ctypes.windll.TcAdsDll  # type: Union[ctypes.CDLL, ctypes.WinDLL]

elif platform_is_linux:
    # try to load local adslib.so in favor to global one
    local_adslib = os.path.join(os.path.dirname(__file__), 'adslib.so')
    if os.path.isfile(local_adslib):
        adslib = local_adslib
    else:
        adslib = 'adslib.so'

    _adsDLL = ctypes.CDLL(adslib)

    LNOTEFUNC = ctypes.CFUNCTYPE(None, ctypes.POINTER(SAmsAddr),
                                 ctypes.POINTER(SAdsNotificationHeader),
                                 ctypes.c_ulong)
```

stlehmann, 2018-08-16:
Yes, that is a known bug which I already fixed in https://github.com/stlehmann/pyads/commit/e2e44bd78a67f006f2c62ebd22ec96dba95f5ac8. Please checkout the current master branch and give it a try. Please send feedback if it's working. Then I will push it on to PyPi.

stlehmann, 2018-08-16:
I just released a new Version 3.0.3 with the bugfix on PyPi.

closed 2018-08-16 by stlehmann

ajaxjjj20, 2018-08-16:
Awesome, thank you @stlehmann for updating that!  It is working now on Windows.