stlehmann/pyads #60: Reading array data

gulyash, 2019-01-17:
Hello, I'm new to using TwinCAT and your library.
On my TC2 device I have an ARRAY [1..160] OF channel, each channel stores variables "Status" (USINT) and "Data" (UINT). My task is to read array data to keep track of changes.
I am currently reading values of the array in a loop, iterating by offset. On each iteration I read status and data from one channel. This is definitely not a good approach as it takes too long to gather all data in the array.
I have seen examples of reading arrays like PLCTYPE_ARR_*type*, but I don't know what to do if I have an array of channels.

stlehmann, 2019-01-18:
Yes reading arrays is as easy as this:

     out = connection.read_by_name("my_array_name", PLCTYPE_INT * 20)

Just multiply the type of your variable with the length of your array. This should work with all types.

stlehmann, 2019-01-22:
Did this solve your issue? Then I would close it.

closed 2019-01-22 by stlehmann