stlehmann/pyads #61: array datatypes for function read_write

bpartzsch, 2019-03-22:
Hi,
I am currently trying to use the function `read_write` on a `Connection` instance. However I figured out that for the write parameter (`plc_write_datatype`) only none-array types are working. It would be very nice to be able to use array types as well.
I assume implementing this is quiet trivial as it is already working fine for the `write` function.
Just copying the folllowing code from `pyads_ex.py` / `adsSyncWriteReqEx` 
```python
    else:
        if type(plc_data_type).__name__ == "PyCArrayType":
            data = plc_data_type(*value)
        else:
            data = plc_data_type(value)
```
.. to `adsSyncReadWriteReqEx2`. But changing `plc_data_type` to `write_data_type`.

PS: Thanks for sharing your great work on this Python package.

stlehmann, 2019-03-22:
Thanks for opening this issue. Would you like to make a Pull-Request to solve this issue?

bpartzsch, 2019-03-25:
I'm not that used to the Git/GitHub workflow but would give it a try. Do you prefer the fork PR approach or via giving me write access?

stlehmann, 2019-03-25:
Please fork the repo and create a PR.

PR #62 (2019-03-26): Update pyads_ex.py
I want to resolve #61. And allow users to use array types for write data on function `readwrite`.
closed 2019-03-26 by stlehmann