stlehmann/pyads #64: "struct.error: unpack requires a buffer of 1 bytes" when trying to add notification

EsbenLorenzen, 2019-03-30:
I have a Twincat 2 project, with UDT's as global variabels.

I was hoping to be able to get notifications when boolean-variables changes, by doing the following:

`plc.connection.add_device_notification('.dStatus.bHardwareOk', pyads.NotificationAttrib(4), cb)`

My callback function looks like this:
```python
@plc.connection.notification(pyads.PLCTYPE_BOOL)
    def callback(handle, name, timestamp, value):
        print('{}: recieved notification for {} :: {}'.format(timestamp, name, value))
```

I get the following error:

> Traceback (most recent call last):
>   File "_ctypes/callbacks.c", line 232, in 'calling callback function'
>   File "C:\Users\Esben L\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyads\pyads_ex.py", line 678, in wrapper
>     return callback(notification, data_name)
>   File "C:\Users\Esben L\AppData\Local\Programs\Python\Python37-32\lib\site-packages\pyads\ads.py", line 737, in func_wrapper
>     bytearray(data)[:data_size]
> struct.error: unpack requires a buffer of 1 bytes

When I add a variable directly, like "MAIN.testbool", it works as expected. Reading the README, I see that the callback notification is not supported for structs.

Is there any way to get around this? Almost all of my variables are places in structs, for.... well for structures sake ;)

Any help and tips would be appreciated! And thank you for the effort put into this project!

EsbenLorenzen, 2019-04-01:
I think I have solved this by myself by creating a NotificationAttribute "by hand":

```python
attr = pyads.NotificationAttrib(sizeof(pyads.PLCTYPE_BOOL), trans_mode=4)
```

An using that when adding my notification:
```python
cb_handles.append(plc.connection.add_device_notification('.dStatus.bHardwareOk', attr, cb))
```

This seems to be working. I guess I was passing the wrong size to the wrapper.

Now in regards to notification handles and how they are disposed, is there a way to "clear" all the added notification?

Or is it a matter of saving the user_handles and deleting them one by one, before stopping the plc connection?
In production the notification setup would only be done once per restart of the machine, however developing, I am constantly reloading the python-script, while the PLC keeps going. Is this causing any "clutter" on the PLC/AMS Router?

pyhannes, 2019-04-04:
I'm facing the same issue as you but.
I'm using 2 connections in a single thread and sometimes this error message comes after the first bunch of notification data comes in.
After restarting the application 1-2 times it suddenly works perfect.

I already made my own callbacks safe with Locks to prevent race conditions but still the same problem. Any ideas?

EsbenLorenzen, 2019-04-05:
I've yet to try my application with a higher number of reads in the same cycle. The most I've tested was around 10 symbols, read every 5 seconds.

I mostly need on-change notifications, but I'll try to test with cyclic reads, to see if I can replicate your issues. Are you using a single callback function for all 30 symbols?
I create a seperate callback for every symbol I need to read. Might not be the right way to do it, but it has worked so far.

Will report back in an hour or so.

pyhannes, 2019-04-05:
Yes in have created 30 on-change notifications (30 different symbols) which’s are all connected to the same callback function. All of the notifications are triggered from a single PLC cycle.

EsbenLorenzen, 2019-04-05:
Did a very quick test, and was able to read 70+ symbols every two seconds using notifications. Different kind of datatypes, with each symbol having it's own callback function.

Using on-change I have added 100+ symbols, without trouble.

Are all the symbols of the same datatype? If not, you will need at least a seperate callback for each type.

pyhannes, 2019-04-05:
Can you also try this while have 2 connections open where the notifications re using the same callback function?

EsbenLorenzen, 2019-04-05:
Two connections as in two seperate PLC's? Or two connections to the same PLC. The latter is not easily achievable, as far as I know, given that the AMS router only supports a single TCP connection to each IP-address. 

It is not possible for me to setup an extra TC connection at the moment, as I have some work to do. I'll try to get around to it this weekend.

Meanwhile, if you share the relevant code, we might be able to pinpoint the issue without having to test it. This way @stlehmann is also able to weigh in, as he has a much deeper understanding of this than I.

pyhannes, 2019-04-05:
It 2 connections to a single PLC but different tasks/ports of it.

EsbenLorenzen, 2019-04-05:
My former comment was based on the discussion in this thread: [https://github.com/Beckhoff/ADS/issues/30](url)

However, the "struct.error: unpack requires a buffer of 1 bytes" message, appears to show up when the callback wrapper expects a different data-length than what is provided. Based on that, my suggestion would be to verify that the correct c-type is passed to the decorator.

Also, how are you choosing which connection to use for the decorator?
From readme example:
```python
>>> @plc.notification(pyads.PLCTYPE_INT)
>>> def callback(handle, name, timestamp, value):
>>>     print(
>>>         '{0}: received new notitifiction for variable "{1}", value: {2}'
>>>         .format(name, timestamp, value)
>>>     )
```
I'm not sure you can use a single callback with two different connections. 
But without seeing how you have set this up, I'm guessing in the wild, which does not seem to be an efficient use of either of our time. So I would need more information, in order to help any further.

stlehmann, 2019-06-06:
@EsbenLorenzen This issue should be fixed with #68. Could you please try if this works for you, now?

closed 2020-04-28 by stlehmann