stlehmann/pyads #66: ADSError: ROUTERERR_PORTALREADYINUSE (1286)

evanmj, 2019-04-04:
Hi,

Thanks for your work on this.  Do you have an insight on to why I'd be getting error 1286?  My code is as follows:
`
pyads.open_port()
remote_ip = '192.168.1.38'
adr = pyads.AmsAddr('5.59.211.2.1.1',851)
pyads.add_route(adr,remote_ip)
plc = pyads.Connection('5.59.211.2.1.1',851)
plc.open()
`

In my case, 192.168.1.38 is the remote machine, with NetID 5.59.211.2.1.1. 

The 'add_route' method returns "connected to 192.168.1.38", but then plc.open() gives the port in use error.

I noticed the route is not in the routing table of the remote PLC... does it work on remote machines instead of 127.0.0.1.1.1?  I don't see a method for adding the password, so I'm not sure how it would work.  To work around that, like I've done with node-ads in the past, I created a static local route to my machine from the PLC side, but I'm not able to connect, given the error above.

I'm on ubuntu with a win7 TC3 PLC.

Thanks!

evanmj, 2019-04-05:
Update:

I poked around a bit, but I'm still having trouble.  The latest is "ADSError: Internal Error (1)".

chrisbeardy, 2019-09-05:
Did you ever get this working, I have just started using and had similar issues, however I have resolved them.

Have you tried running 

`python
import pyads
>>> SENDER_AMS = '1.2.3.4.1.1'
>>> PLC_IP = '192.168.0.100'
>>> USERNAME = 'user'
>>> PASSWORD = 'password'
>>> ROUTE_NAME = 'RouteToMyPC'
>>> HOSTNAME = 'MyPC'
>>> PLC_AMS_ID = '11.22.33.44.1.1'
>>> pyads.add_route_to_plc(SENDER_AMS, HOSTNAME, PLC_IP, USERNAME, PASSWORD, route_name=ROUTE_NAME)
`
from linux to add the route, username and passwork is the PLC.

Then using the follwing to access vars:

'pyhton
import pyads
pyads.open_port()
pyads.set_local_address('add.used.ear.lier')
pyads.close_port()

plc = pyads.Connection('plc_ads', port, 'plc_ip')
plc.open()
'

on linux i get a warning about a port being out of range however it still works. 

you can also use the with statement instead of plc = pyads.Connection.

blindidiotscout, 2019-09-18:
Hi,
I have the same issue as described by evanmj. I was able to add the route as you described. Further your second statement did not produce an error after plc.open(). plc.is_open gave me a true but the connection always closes automatically after a view seconds.

> 2019-09-18T14:37:49+0200 Info: Connected to 192.168.66.27
> 2019-09-18T14:38:09+0200 Info: connection closed by remote

Even after the connection is closed I get a true for plc.is_open. Other commands like plc.read_device_info() still produce an error.
 

> Traceback (most recent call last):
>   File "<input>", line 1, in <module>
>   File "/usr/local/lib/python3.7/dist-packages/pyads/ads.py", line 540, in read_device_info
>     return adsSyncReadDeviceInfoReqEx(self._port, self._adr)
>   File "/usr/local/lib/python3.7/dist-packages/pyads/pyads_ex.py", line 374, in adsSyncReadDeviceInfoReqEx
>     raise ADSError(error_code)
> pyads.pyads_ex.ADSError: ADSError: timeout elapsed (1861). 

Do you have any suggestions?

chrisbeardy, 2019-09-18:
I had similiar issues too, I'm not 100% sure what situation gave me teh same problem as you All i know is that I get a working method (albeiit with warnings) after doing what I described above. Have you tried using the ```with``` statement? 

The linux ADS router I find isn't perfect and can take some fiddling to get something working but overall it works well.

blindidiotscout, 2019-09-18:
I just got it to work two minutes ago! What did the trick for me was altering the add_route_to_plc command. The documentation says something like:
`pyads.ads.add_route_to_plc(sending_net_id, adding_host_name, ip_address, username, password, route_name=None, added_net_id=None)`
I filled out all the fields and switched Hostname and Routename, resulting in:
```
pyads.add_route_to_plc('192.168.66.53.1.1', 'Pi', '192.168.66.27', 'Administrator', '', route_name='192.168.66.53',added_net_id='192.168.66.53.1.1')
plc = pyads.Connection('5.18.105.98.1.1', pyads.PORT_SPS1, '192.168.66.27')
plc.is_open
False
plc.open()
2019-09-18T17:02:41+0200 Info: Connected to 192.168.66.27
plc.read_by_name('MAIN.Logger_OnOFF', pyads.PLCTYPE_BOOL)
True
```

chrisbeardy, 2019-09-18:
yes it can be a little confusing sometimes which way aound the IP addresses etc should beweenn the PLC and the route. But great that you got it.

Auvite, 2019-10-07:
Struggling with the same issue. 
Just to make sure from the latest blindidiotscout example

> pyads.add_route_to_plc('192.168.66.53.1.1', 'Pi', '192.168.66.27', 'Administrator', '', >route_name='192.168.66.53',added_net_id='192.168.66.53.1.1')
> plc = pyads.Connection('5.18.105.98.1.1', [...]
 
sending_net_id= linux address(AMS)
adding_host_name= linux hostname
ip_address= linux ip
username= linux user
password=linux pw
route_name=can be picked as wanted
added_net_id= linux address(AMS)

Correct?

BR
TAu

chrisbeardy, 2019-10-07:
sender_net_id = linux_ams
hostname = linux hostname
ip_address = PLC_IP address 
username = PLC user name
password = PLC password
route_name = anything you want
 
pyads.add_route_to_plc(SENDER_AMS, HOSTNAME, PLC_IP, USERNAME, PASSWORD, 
    route_name=ROUTE_NAME)

See if this works?

chrisbeardy, 2019-10-07:
you only need to do teh above once from a seperate script then in teh future when using the program you can do 

```
import pyads
pyads.open_port()
pyads.set_local_address('address.used.ear.lier')
pyads.close_port()
plc = pyads.Connection('plc_ads', port, 'plc_ip')
plc.open()
```

Auvite, 2019-10-07:
For some reason doesn't get the "port already in use" error anymore but "timeout elapsed(1861)"

The route adding looks like ok from plc side. 
plc.is_open is false
after plc.open() 
Info: Connected to 10.xxx.x.x
Info: Connection closed by remote.
and after this last lines referes to timeout.

Totally confused.

chrisbeardy, 2019-10-07:
now you have the route set up OK?

Did you use:

```
AMS_Address = AMS address of linux (sender_net_id)
pyads.open_port()
pyads.set_local_address(AMS_Address)
pyads.close_port()
```

before plc.open()

Auvite, 2019-10-08:
Yes i believe route is ok. PLC has 
[image]

And the code

>import pyads
>
>pyads.open_port()
>ams='10.114.2.4.1.1'
>pyads.set_local_address(ams)
>pyads.close_port()
>port=851
>plc = pyads.Connection('5.56.56.35.1.1', port, '10.114.2.3')
>plc.open()
>VAR=plc.read_by_name('GVL.distance', pyads.PLCTYPE_INT)
>print(VAR)
>plc.close()

Resulting
>2019-10-08T10:48:10+0300 Info: Connected to 10.114.2.3
>2019-10-08T10:48:12+0300 Info: connection closed by remote
>Traceback (most recent call last):
>  File "/home/pi/test/test.py", line 10, in <module>
>    VAR=plc.read_by_name('GVL.distance', pyads.PLCTYPE_INT)
>  File "/home/pi/.local/lib/python3.7/site-packages/pyads/ads.py", line 830, in read_by_name
>    handle=handle,
>  File "/home/pi/.local/lib/python3.7/site-packages/pyads/pyads_ex.py", line 695, in >adsSyncReadByNameEx
>    handle = adsGetHandle(port, address, data_name)
>  File "/home/pi/.local/lib/python3.7/site-packages/pyads/pyads_ex.py", line 658, in adsGetHandle
>    PLCTYPE_STRING,
>  File "/home/pi/.local/lib/python3.7/site-packages/pyads/pyads_ex.py", line 542, in >adsSyncReadWriteReqEx2
>    raise ADSError(err_code)
>pyads.pyads_ex.ADSError: ADSError: timeout elapsed (1861). 
>>>>

chrisbeardy, 2019-10-08:
@Auvite i believe you have found a BUG! However I have just done a quick test and found a workaround but this needs sorting.

Try making the route like this:

```
import pyads
SENDER_AMS = '192.168.27.2.1.1'
PLC_IP = '192.168.23.4'
USERNAME = 'plc_user'
PASSWORD = 'plc_password'
# ROUTE_NAME = 'PI'  # there issue appears when you pass this in
HOSTNAME = 'RASPBERRY'
PLC_AMS_ID = '192.168.27.1.1.1'
pyads.add_route_to_plc(SENDER_AMS, HOSTNAME, PLC_IP, USERNAME, PASSWORD)
```
There appears to be a bug that if you pass the route name in, it using this route_name as the 'address' instead. 

What version of TwinCat are you using? They may have changed the router interface.

Auvite, 2019-10-08:
Version details:

[image]

Auvite, 2019-10-08:
Bon appétit

>import pyads
>
>AMS='LINUX AMS'
>PLC='PLC IP'
>USER='PLC_USERNAME'
>PW='PLC_PASSWORD'
>HOSTNAME='LINUX_IP'
>PLC_AMS_ID='PLC_AMS'
>pyads.add_route_to_plc(AMS,HOSTNAME,PLC,USER,PW)

This seems to work with my setup.

blindidiotscout, 2019-10-08:
That is exactly the point I was trying to make! I already had a conversation with @stlehmann on that bug.

chrisbeardy, 2019-10-08:
> That is exactly the point I was trying to make! I already had a conversation with @stlehmann on that bug.

Sorry i missed this first time

blindidiotscout, 2019-10-08:
No, problem. In retrospective I wasn't very clear in my post.

blindidiotscout, 2019-10-08:
If adding the route was correct it should look something like this:
[image]
In my case I derived the AMS of the Pi from its static ip and set it using:
`import pyads`
`pyads.open_port()`
`pyads.set_local_address('192.168.66.53.1.1')`
Then I added the route like this:
`SENDER_AMS = '192.168.66.53.1.1'`
`PLC_IP = '192.168.66.27'`
`USERNAME = 'Administrator'`
`PASSWORD = ''`
`ROUTE_NAME = '192.168.66.53'`
`HOSTNAME = 'Pi'`
`PLC_AMS_ID = '5.18.105.98.1.1'`
`pyads.add_route_to_plc(SENDER_AMS, HOSTNAME, PLC_IP, USERNAME, PASSWORD, route_name=ROUTE_NAME, added_net_id=SENDER_AMS)`
After adding the route and seeing something similar to above you can connect to the pi. Don't forget, that your password and username are different with TwinCAT 3! Connect via:
`plc = pyads.Connection('5.18.105.98.1.1', 801, '192.168.66.27')`
`plc.open()`
Producing something like:

> 2019-10-08T15:17:58+0200 Info: Connected to 192.168.66.27

Then test it with:
`plc.read_device_info()`
getting something like:

>  ('TCatPlcCtrl', <pyads.structs.AdsVersion object at 0x75b68a70>)

I hope this helps...

stlehmann, 2019-10-08:
Maybe @Adrian-at-CrimsonAuzre who implemented this feature could have a closer look at this?

Adrian-at-CrimsonAuzre, 2019-10-08:
I implemented it on TwinCAT2 because that's all I had access to, so I don't know how TwinCAT 3 might be different.

It looks like maybe TwinCAT3 treats the route name differently than TwinCAT 2 does so removing that functionality *probably* won't hurt anything, but it should be tested against a live system before being merged though.

chrisbeardy, 2019-10-08:
can someone with TWINCAT 2 test #97 and #98 (or another solution) to see if theres a solution that works for TC2 and TC3.

otherwise I think the easiest way to clear up these issues may just be by clarifying the function in the documentation?

blindidiotscout, 2020-01-14:
Did anyone of you encounter the problem of methods working in the python shell but not when executing a script? I can read_by_name and write_by_name (example only for reading):

`import sys`
`print(sys.version)`
`3.7.3 (default, Apr  3 2019, 05:39:12) `
`[GCC 8.2.0]`
`import pyads`
`plc = pyads.Connection('5.18.105.98.1.1', 801, '192.168.66.37')`
`plc.open()`
`2020-01-14T08:02:42+0100 Info: Connected to 192.168.66.37`
`i = plc.read_by_name('MAIN.Logger_OnOFF', pyads.PLCTYPE_BOOL)`
`print(i)`
`True`
`plc.close()`
`2020-01-14T08:03:56+0100 Info: connection closed by remote`

If I execute following code:

`import sys`
`import pyads`
`print(sys.version)`
`plc = pyads.Connection('5.18.105.98.1.1', 801, '192.168.66.37')`
`plc.open()`
`if plc.is_open:`
`    i = plc.read_by_name('MAIN.Logger_OnOFF', pyads.PLCTYPE_BOOL)`
`    print(i)`
`plc.close()`

I get:

```
runfile('/tmp/pycharm_project_97/ADS.py', wdir='/tmp/pycharm_project_97')
3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0]
2020-01-14T08:11:55+0100 Info: Connected to 192.168.66.37
Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "/home/pi/.pycharm_helpers/pydev/_pydev_bundle/pydev_umd.py", line 197, in runfile
    pydev_imports.execfile(filename, global_vars, local_vars)  # execute the script
  File "/home/pi/.pycharm_helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/tmp/pycharm_project_97/ADS.py", line 8, in <module>
    i = plc.read_by_name('MAIN.Logger_OnOFF', pyads.PLCTYPE_BOOL)
  File "/home/pi/virtualenvs/cv/lib/python3.7/site-packages/pyads/ads.py", line 830, in read_by_name
    handle=handle,
  File "/home/pi/virtualenvs/cv/lib/python3.7/site-packages/pyads/pyads_ex.py", line 695, in adsSyncReadByNameEx
    handle = adsGetHandle(port, address, data_name)
  File "/home/pi/virtualenvs/cv/lib/python3.7/site-packages/pyads/pyads_ex.py", line 658, in adsGetHandle
    PLCTYPE_STRING,
  File "/home/pi/virtualenvs/cv/lib/python3.7/site-packages/pyads/pyads_ex.py", line 542, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: timeout elapsed (1861). 
```

Any ideas on why this is happening?

Auvite, 2020-01-14:
First thought is bad network connection or some other delay. Does this
happen everytime or time to time?

blindidiotscout, 2020-01-15:
Thank you very much, it was the connection quality! Using Ethernet instead of Wifi did the trick.

stlehmann, 2020-04-27:
Can this issue be closed?

closed 2020-04-27 by stlehmann