stlehmann/pyads #71: PLC Connection Linux

Adrian-at-CrimsonAuzre, 2019-07-11:
I'm trying to connect to a Beckhoff PLC using this library on Linux and I cannot get a connection to hold. The code:
`with pyads.Connection('5.13.82.20.1.1', pyads.PORT_SPS1, '192.168.200.211') as plc:
	return plc.read_by_name('.AUTO_WITH_PARTS', pyads.PLCTYPE_BOOL)`
On Windows this runs flawlessly, but on Ubuntu it displays:
`2019-07-11T10:01:01-0400 Info: Connected to 192.168.200.211
2019-07-11T10:01:06-0400 Info: connection closed by remote`

When I don't specify the IP address it sits there for a few minutes and returns:
`2019-07-11T10:06:51-0400 Error: Connect TCP socket failed with: 110`

Do I still need to add my Ubuntu PC's AMS address to the PLC manually as discussed [here](https://github.com/stlehmann/pyads/issues/34#issuecomment-331694365)?

Output from setup.py test
`----------- coverage: platform linux, python 3.6.9-final-0 -----------
Name                  Stmts   Miss  Cover
-----------------------------------------
pyads/__init__.py        10      0   100%
pyads/ads.py            178     15    92%
pyads/constants.py      123      4    97%
pyads/errorcodes.py       1      0   100%
pyads/filetimes.py       21      0   100%
pyads/pyads_ex.py       234     38    84%
pyads/structs.py         93     15    84%
pyads/utils.py            7      0   100%
-----------------------------------------
TOTAL                   667     72    89%`

stlehmann, 2019-07-11:
Yes you always need to add the route to your PLC and vice versa manually as described in the given link.

Adrian-at-CrimsonAuzre, 2019-07-11:
I would honestly have no idea how to implement this, but after poking around with Wireshark I semi-reverse engineered the process for adding a new route on a remote PLC. Anyone else want to take a stab at it?
[image]
This is after adding routes to 4 different Bekhoff PLCs, all of the bytes labeled "protocol stuff" don't seem to change so I'm guessing they're setting flags that are constant?

Adrian-at-CrimsonAuzre, 2019-07-12:
Well I took a stab at it, see #75

stlehmann, 2019-07-12:
Adding routes is already implemented. Use pyads.add_route() for it. It should actually work just fine. I don't get the point of your reverse-engineering and PR #75. Could you explain which problem you try to tackle with this.

Adrian-at-CrimsonAuzre, 2019-07-12:
With PR #75 you can add your AMS ID to the PLC without needing to add it in Twincat first. In my example, I connected to the PLC with an ID that I have never used before, and never added to my PLC with Twincat.

Adrian-at-CrimsonAuzre, 2019-07-30:
Pull #75 merged, issue resolved.

closed 2019-07-30 by Adrian-at-CrimsonAuzre