stlehmann/pyads #76: How to get notification to work on Linux?

dczanella, 2019-07-12:
Hi

I'm running the test as described [here](https://github.com/stlehmann/pyads) to handling notification callbacks on my Debian 9 Linux. I run the testserver on separate screen that showed that the notifications was added, but I the callback function has never been called.
I checked a lot, but I couldn't find the mistake.

On first ipython screen

import pyads
from pyads.testserver import *
server = AdsTestServer(handler=AdvancedHandler())
server.start()

On second ipython screen

import pyads
plc = pyads.Connection('127.0.0.1.1.1', 48898)
plc.open()
@plc.notification(pyads.PLCTYPE_INT)
def callback(handle, name, timestamp, value):
    print(
        '{1}: received new notitifiction for variable "{0}", value: {2}'
        .format(name, timestamp, value)
    )
handles = plc.add_device_notification('GVL.intvar',
                                          pyads.NotificationAttrib(2), callback)
# Write to the variable to trigger a notification
plc.write_by_name('GVL.intvar', 123, pyads.PLCTYPE_INT)

In first ipython terminal I have:

INFO:Command received: READWRITE (index group=61443, index offset=0, read length=4, write length=11, write data=GVL.intvar)
INFO:Command received: ADD_DEVICE_NOTIFICATION
INFO:Command received: READWRITE (index group=61443, index offset=0, read length=4, write length=11, write data=GVL.intvar)
INFO:Command received: WRITE (index group=61445, index offset=0, data length=2, value={
INFO:Command received: WRITE (index group=61446, index offset=0, data length=4, value=

But in second I cannot see the callback runs.
Thanks

chrisbeardy, 2019-09-05:
I know this is a little out of date and I know this is probably not useful, but I've had a look at your code and tried it on my machine, it runs fine which is interesting, have you tried running using normal python terminals (command line) instead of ipython?

I find the linux implementation can be really temperatmental, and can behave differnetly distro to distro and depending whats installed on the distro.

stlehmann, 2020-04-28:
Is this issue fixed?

closed 2020-04-28 by stlehmann