stlehmann/pyads #77: NetBIOS timeout on Ubuntu

Adrian-at-CrimsonAuzre, 2019-07-15:
When I make a connection to a specific PLC (don't know the model offhand) it queries my computer's NetBIOS name directly. Ubuntu denies this request, and then the PLC loops through its saved hostnames broadcasting and waiting for an answer which my PC eventually responds to. Trouble is, even when I set the timeout to 30 seconds it still "times out".

```
2019-07-15T08:58:51-0400 Info: Connected to 192.168.200.210
2019-07-15T08:59:02-0400 Error: read frame failed with error: 104
2019-07-15T08:59:02-0400 Info: connection closed by remote
```
That read error happens when the PLC sends a TCP message to my computer acknowledging its existence and pyads (or the underlying ADS library) closes the connection.
[image]

Is there a way to make Ubuntu respond to these direct NetBIOS requests? If not, then is there a way to prevent the connection from closing when there is a read frame error?

Adrian-at-CrimsonAuzre, 2019-07-16:
This didn't solve the issue, but it _was_ the cause of the issues I was having on my Ubuntu Server install.
NetBIOS support is needed on Linux distributions, this is included in most desktop distros but it's not a given on server distros. For example, Ubuntu Server needs `avahi-daemon` in order to correctly read tags from a PLC.

kryskool, 2019-07-17:
@Adrian-at-CrimsonAuzre 

Can you test tips found [here](https://www.linuxquestions.org/questions/linux-networking-3/ping-netbios-names-from-linux-samba-271336/)

edit your **/etc/nsswitch.conf** file
search for the line starting with **hosts** (my line was like this "hosts: files dns")
add **wins** at the end of the line ("hosts: files dns wins")

It can be necessary to install **winbinds**

```
apt-get install winbinds
```

regards,

Adrian-at-CrimsonAuzre, 2019-07-17:
Still doesn't work. I can ping my hostname from Windows just fine and I can ping Windows computers in Ubuntu, but my Ubuntu PC actively refuses the connection when sent a blank query.
This PLC is running the oldest version of TwinCAT out of all our Beckhoffs, maybe updating it would help?

Adrian-at-CrimsonAuzre, 2019-07-23:
Moved my project from an Ubuntu Server development VM to the production CentOS VM and all is well? I'm pretty sure Samba on the CentOS VM is the only difference between the two environments. I'm going to chalk it up to that and close this.

closed 2019-07-23 by Adrian-at-CrimsonAuzre