stlehmann/pyads #80: Saving device notification callback handle makes wrong assumptions

dxpke, 2019-07-26:
Hey,

when using device notifications, you save the handles to the callback functions here:

https://github.com/stlehmann/pyads/blob/af76fd5/pyads/pyads_ex.py#L692

This is correct in the sense that it is required to avoid garbage collection of the handles, leading to crashes. What is not correct is that they are indexed in the handle array only by the notification handle. When using notifications on multiple AMS netids or ports, TcAdsDll may give back the same notification handle twice. The handles seem to be scoped by netid/port (which makes sense since they are determined by the ADS server you query). If this happens, the prior callback gets overwritten and Python crashes (this took a long time to find...). It is quite likely (read: almost guaranteed) to happen when you have newly started an XAR and register multiple notifications on multiple tasks/ports.

There is an easy solution to this, which is saving the callback indexed by netid, port, and notification handle. And additionally checking the callback array so that it does not get overwritten by some accident.

But I think that the current way of handling notification handles is a bit weird anyway. As "user handle", you get back the handle of the variable resolved by `ADSIGRP_SYM_HNDBYNAME`, which you can't really use anyway. In my opinion, the library should take care of remembering that handle to free it later when deleting the notification, and the user does not even need to see it. So a different solution could be to save a structure per netid/port/handle tuple that contains both the variable handle (to free it later) and a reference to the callback function (so it does not get garbage-collected). That would modify the interface of `add_device_notification` however.

Please tell me what you think so I can implement the solution that is more likely to be accepted as PR.

stlehmann, 2019-12-03:
Thanks for the PR. I have never connected to more then one ADS server. So this is an interesting issue and I can see where you' re coming from. I think the solution you gave here makes sense. I will give it a closer look.

dxpke, 2020-01-13:
Any news? Can I help in some way?

stlehmann, 2020-04-28:
I merged your PR which solves this issue. So I guess this issue can be closed.

closed 2020-04-28 by stlehmann