stlehmann/pyads #81: Support for asyncio?

danieljfarrell, 2019-08-15:
It may not be feasible, but are they plans or is it already possible to use a asynchronous transport with pyads. Ideally, so it can be used with asyncio and the async await way of doing things? Sorry if this is the wrong location for a question.

chrisbeardy, 2019-09-05:
did you manage to try this?

danieljfarrell, 2019-09-05:
No, I didn’t make an progress. Would be interesting to get some advice.

chrisbeardy, 2019-09-05:
sorry I've not actually tried this, was interested to see if you had then got results. I'm developing an application at the moment using this library so may give async a go and I'll let you know

stlehmann, 2020-04-28:
It is not planned to implement any async features at the moment. I think if we would do it it should be a separate module.

stlehmann, 2020-11-04:
Actually pyads is implemented in an async way in the [homeassistant integration](https://www.home-assistant.io/integrations/ads/). This could be a nice start point for further development of this feature

chrisbeardy, 2020-11-05:
I think you're right when you say it should be a different module, much like having `requests` and `aiohttp`. As to make it async you would need all of the public methods of pyads to become async, and I think you would want to give users the option of not having it async too. Using pyads in an async application is fairly straightforward but the actual pyads calls are still blocking, this looks similar to what home assistant is doing. e.g.

Current pyads in use in an application which has awaitable tasks (shown with sleep):
```python
import pyads
import time
import asyncio

async def count(plc):
    print("One")
    read = plc.read_by_name("MAIN.bVar", pyads.PLCTYPE_BOOL)
    print(read)
    print("Two")
    await asyncio.sleep(1)
    print("Three")

async def async_main(plc):
    await asyncio.gather(count(plc), count(plc), count(plc))

def main():
    plc = pyads.Connection("192.168.0.1.1.1", 851)
    plc.open()
    s = time.perf_counter()
    loop = asyncio.get_event_loop()
    try:
        loop.run_until_complete(async_main(plc))
    finally:
        loop.close()
    plc.close()
    elapsed = time.perf_counter() - s
    print(f"{__file__} executed in {elapsed:0.2f} seconds.")

main()
```
gives output:
```
One
False
Two
One
False
Two
One
False
Two
Three
Three
Three
```
or to enphasise it's really just blocking, so not actually beneifcial to use async currently in an application:
```python
async def count_german():
    print("ein")
    print("zwei")
    print("drei")

async def count(plc):
    print("One")
    read = plc.read_by_name("MAIN.bVar", pyads.PLCTYPE_BOOL)
    print(read)
    print("Two")
    print("Three")

async def async_main(plc):
    await asyncio.gather(count(plc), count_german())

def async_sample_main():
    plc = pyads.Connection("192.168.0.1.1.1", 851)
    plc.open()
    s = perf_counter()
    #asyncio.run(async_main())
    loop = asyncio.get_event_loop()
    try:
        loop.run_until_complete(async_main(plc))
    finally:
        loop.close()
    plc.close()
    elapsed = perf_counter() - s
    print(f"{__file__} executed in {elapsed:0.2f} seconds.")

async_sample_main()
```
gives:
```
One
False
Two
Three
ein
zwei
drei
```
Making a pyads asyncio library is probably not trivial, making the public methods (e.g. read_by_name) all have async def may be a start, then follow through all the errors....

I guess the idea is having the public methods aysync would allow you do other stuff while pyads is doing its thing e.g.

```python
async def count_german():
    print("ein")
    print("zwei")
    print("drei")

async def count(plc):
    print("One")
    read = await plc.read_by_name("MAIN.bVar", pyads.PLCTYPE_BOOL)
    print(read)
    print("Two")
    print("Three")

async def async_main(plc):
    await asyncio.gather(count(plc), count_german())

```

that could then give you:
```
One
ein
zwei
drei
Two
Three
```

Again not sure how easy this would be to implement, I see some use cases for example, if you are reading/writing large amounts of individual vars using `read_by_name` and don't want to block up the program, although this has been made simpler with the new ads read and write by list and the actual delay in individual ADS calls is not that much, ~10ms

Or potentially reading/writing to multiple PLCs could be a benefit to having it async too. Not sure if the underlying ADS DLL would be able to cope with that though.

stlehmann, 2021-12-10:
There will be no support for asyncio on pyads in the foreseeable future. So I will close this issue.

closed 2021-12-10 by stlehmann