stlehmann/pyads #87: Linux - Warning: Port 0x2710 is out of range, No Response Pending

chrisbeardy, 2019-09-05:
When running on linux using the Connection class:

```python
import ads
import time

pyads.open_port()
pyads.set_local_address('192.168.27.2.1.1')
pyads.close_port()

plc = pyads.Connection('192.168.27.1.1.1', 851, '192.168.23.3')
plc.open()

try:
    while True:
        temp = plc.read_by_name('MAIN.rVar', pyads.PLCTYPE_LREAL)
        print(temp)
        time.sleep(1)
finally:
    plc.close()
```

I have great success, but periodically I get the following Warnings printed in my console:

```
2019-09-05T15:32:49+000 Warning: Port 0x2710 is out of range
2019-09-05T15:32:49+000 Warning: No response pending
```

This doesn't really bother me but has anyone else seen this? know what could be causing it?

stlehmann, 2019-10-03:
I just got annoyed by the same warnings. The cause is in the adslib. I found a closed issue to the topic https://github.com/Beckhoff/ADS/issues/89. But I don' t really know the specific reason for it.

stlehmann, 2019-10-03:
The warning stops if I close TwinCAT on the remote PC. Seems like it is sending on the same port.

chrisbeardy, 2019-10-03:
> The warning stops if I close TwinCAT on the remote PC. Seems like it is sending on the same port.

interesting, you mean you closed visual studio on the PLC? I shall give this a go.

stlehmann, 2019-10-03:
Yes this worked for me.

chrisbeardy, 2019-10-04:
This seems to work for me to. seems its the scope server.

Closed: Close Twincat IDE any message goes away.

closed 2019-10-04 by chrisbeardy