stlehmann/pyads #93: Is it possible to access an ADS device without an intermediate PLC?

Kamforka, 2019-09-29:
I wonder if it's possible to directly send instructions to an `EK1100` bus coupler and using the `pyads` package without having a PLC in between the PC and the bus coupler?

stlehmann, 2019-09-30:
pyads does not support auto-discover.

closed 2019-10-04 by stlehmann

jugge83, 2020-09-09:
@stlehmann I to am very interested in knowing if it is possible to connect directly to a bus coupler/slave unit without any additional "brokers" (masters, TwinCat etc.). In my case I plan to use an EK1101.

Kamforka, 2020-09-10:
@jugge83 it is possible, I've done it myself using this lib.

jugge83, 2020-09-10:
@Kamforka Thank you for the quick reply!  I couldn't find any information on it. 
Do you have any ball park estimates on responce/cycle times for data/input reading with a coupler?

Kamforka, 2020-09-10:
Well the cycle time will depend on your program execution time. I've basically set up a while loop and a sleep of 50ms, so that way I could fetch new data every 50ms obviously. 
I guess there is a maximum threshold for the EK coupler in terms of cycle time, in case you do a read/write cycle every 10s I'm sure it will error out, but if you keep it low like <100ms you should be good to go.
Also check out the examples in the repo, I used to work my way out of them.

jugge83, 2020-09-10:
I usually create some kind of adaptive reconnect on errors so long cycle times shouldn't be an issue in my application.
When controlling modbus couplers (wago) I usually get around 30-50 ms response times when reading coils and registers. I think that most of the delay is due to some cycle time that comes from the data translation to and from the bus. 
From what I've read EtherCat should be at least an order of magnitude faster. 
Thanks for the Example tip!

Kamforka, 2020-09-10:
Oh yeah the coupler should be in the microsecond domain, but you should check that in the manual. Reading and writing with pyads should be pretty fast as well, since it's just a Cython wrapper around the SOEM package which is written in pure C.