stlehmann/pyads #104: Can't add a route with windows

adrien-zinger, 2019-11-18:
Hi !

I run a server on my host (windows) and I can't add a route with the Twincat Ams remote manager. The twincat dialog always returns ADS error 13 (the port is not connected) and I can't find any explication.

I launched a docker container from your Dockerfile I can perfectly add a route with the add_route function and it works! I also tried to run the server in the container but nothing to do, I always can't add route from windows.

Is there another method to add a route on windows? Or have you already seen this error 13?

Thank you

stlehmann, 2019-11-19:
Hi,

add_route is only supported for Linux clients. For Windows you need to use TwinCAT to create
routes.

Regards, Stefan.

adrien-zinger, 2019-11-20:
Hi,

Thank you for your reply.

Yes of course as it's writed in your documentation, but the Add Route Dialog of twincat can't see the server. I tried everything with the others twincat3 softwares. The dialog keep returning ADS error 13 and the bekhoff documentation is not really exhaustive... Did you already spot this error before? Can we connect to the testserver from windows?

Cordially, Adrien

stlehmann, 2019-11-23:
The testserver is just for automated testing. You shouldn't be using it for your projects. If you have TwinCAT installed on your system then the easiest way to test ADS functionality is to start a new Twincat project and use pyads to connect to the plc.

closed 2019-12-05 by stlehmann