stlehmann/pyads #105: Could not find module 'TcAdsDll' with python 3.8.0

yannickasselin, 2019-11-22:
Hi,
Running pyads in python 3.7.5 works fine but when trying to import pyads in python 3.8.0 I get this error:

FileNotFoundError: Could not find module 'TcAdsDll'. Try using the full path with constructor syntax.

klauer, 2019-11-22:
Python 3.8 changed how Windows DLL paths work with ctypes, only loading from trusted locations.

See:
https://docs.python.org/3/whatsnew/3.8.html#ctypes
https://docs.python.org/3/whatsnew/3.8.html#changes-in-the-python-api (search for DLL)

yannickasselin, 2019-11-22:
I'm sorry. I'm really just a beginner. What do I need to do to make it work?

stlehmann, 2019-11-23:
OK, seems we need to use [add_dll_directory](https://docs.python.org/3/library/os.html#os.add_dll_directory) here to allow loading the DLL from working directory.

closed 2020-04-03 by stlehmann

molinerozadkiel, 2021-06-11:
Hi, I'm having this same problem, using Windows, python version is: 3.9.5, and pyads version is 3.3.7
what should I do to allow the usage of the dll file?? I also ran a search throughout the whole OS for "TcAdsDll.dll" and got no result.

any advise?

stlehmann, 2021-06-12:
Hey Sofia, did you install TwinCAT on Windows?

miklg, 2023-05-22:
Hi @stlehmann It's 100% necessary to have installed TwinCAT in Windows to communicate with Beckhoff PLC via pyads?

chrisbeardy, 2023-05-22:
@miklg Yes it is, or at least just the ADS version, see my stack opverflow answer: https://stackoverflow.com/questions/76305160/windows-10-python-pyads-library-error-could-not-find-module-tcadsdll-dll