stlehmann/pyads #109: adserror server not started (6) target port not found

amarco, 2019-12-24:
I am at a loss. From my XAEshell enviroment I can connect to my plc and download code, debug code rune code on the plc. But when I try to access the plc from pyads from the same windows computer as xae I get an adserror about server not started and port not found. I see I am connectd to the plc from within routes on the windows pc. 
What am I missing?

Marco

kryskool, 2019-12-26:
Hi @amarco 

can you show us your code ?

regards,

amarco, 2019-12-26:
Code is really basic and based on examples.

import pyads

plc=pyads.Connection('10.10.32.225.1.1',pyads.PORT_SPS1)
plc.open()

#this returns true
print("Opened PLC {0}".format(plc.is_open))

#this fails
info=plc.read_device_info()
print("PLC Info {0}".format(info))

#this fails as well
fastFrameBufferReady = plc.read_by_name("GVL.ioList.fastFrameBufferReady",pyads.PLCTYPE_BOOL)
print("FastFrameBufferReady {0}".format(fastFrameBufferReady))

amarco, 2019-12-29:
after exploring further I started to look at why it would not communicate so I was looking at port number and noticed that PORT_SPS1 is for TwinCat 2 and I am using TwinCat 3 so I change to PORT_TC3PLC1 and things started to work. I also then had to remove the reference to GVL to get the read to actually return a value.

stlehmann, 2019-12-30:
So this issue can be closed?

closed 2019-12-30 by amarco

amarco, 2019-12-30:
Yes I closed the ticket, sorry for that..

RWDI - A Platinum Member of Canada's 50 Best Managed Companies
This communication is intended for the sole use of the party to whom it was addressed and may contain information that is privileged and/or confidential. Any other distribution, copying or disclosure is strictly prohibited. If you received this email in error, please notify us immediately by replying to this email and delete the message without retaining any hard or electronic copies of same. Outgoing emails are scanned for viruses, but no warranty is made to their absence in this email or attachments. If you require any information supplied by RWDI in a different format to facilitate accessibility, contact the sender of the email, email solutions@rwdi.com or call +1.519.823.1311.
Please be aware that when you contact us with a business query we may collect and use your details for future communications.