stlehmann/pyads #116: Timestamp precision in device notifications

dxpke, 2020-03-02:
Hi,

I noticed that in the device notification decorator, the timestamp from the ADS packet (which is a `FILETIME` with precision of 100 ns) is converted to a Python `datetime` (which has a precision of 1 µs). This means that some precision is lost in the resulting timestamps. The TwinCAT system can run with a tick base time of 62.5 µs, for example, which would mean that values cyclically acquired from a corresponding task would have inexact and non-equidistant timestamps.

I was wondering how this situation could be resolved, maybe by adding a parameter to the notification decorator so that it does not touch the timestamp?

koppa, 2020-04-20:
I would suggest a `return_ctypes` argument for `add_device_notification` which does not convert the data types automatically to native python types.

This solves also the problem that the returned datatype for notification data is not consistent.

stlehmann, 2020-04-28:
> I was wondering how this situation could be resolved, maybe by adding a parameter to the notification decorator so that it does not touch the timestamp?

That would be a good option. Maybe a parameter called `use_filetime` could be used to switch between datetime and filetime.

stlehmann, 2020-04-28:
@koppa 

> I would suggest a return_ctypes argument for add_device_notification which does not convert the data types automatically to native python types.

That would be a complete change in pattern. pyads wants to provide a highlevel interface. Working with ctypes doesn't fit well in this. Also this pattern would need to be applied to other functions as well for consistency.

stlehmann, 2020-05-02:
@koppa After having another look at the codebase I've seen that we already applied the return_ctypes argument in some functions. So this might be a thing to go for after all. Would you like to create a PR?

closed 2020-05-04 by stlehmann

koppa, 2020-05-05:
@stlehmann 
Sure, I will create a PR in the next days.

dxpke, 2020-05-05:
Feature is already implemented.

-- 
Sitz der Gesellschaft: Ilmenau
Registergericht: Amtsgericht Jena, HRB 
505629
Geschäftsführer: Kazuya Hayafune, Yujin Okochi, Michael Löffler

USt-IdNr.: DE272427674
Datenschutzerklärung : 
https://drivexpert.de/de/datenschutzerklaerung/ 
<https://drivexpert.de/de/datenschutzerklaerung/>
Information nach Art. 13 
/ Art. 14 DS-GVO über die Verarbeitung Ihrer Daten 
und Ihre Rechte nach 
der EU-Datenschutz-Grundverordnung : https://drivexpert.de/?p=629 
<https://drivexpert.de/?p=629>

Diese E-Mail (inklusive deren Anlagen) 
enthält vertrauliche und/oder rechtlich geschützte
Informationen. Wenn Sie 
nicht der richtige Adressat sind oder diese E-Mail
irrtümlich erhalten 
haben, informieren Sie bitte sofort den Absender und
vernichten Sie diese 
Mail. Das unerlaubte Kopieren sowie die unbefugte
Weitergabe dieser Mail 
ist nicht gestattet.
Vielen Dank.

This e-mail (including all attached 
documents) may contain confidential and/or privileged information. If you 
are not the intended recipient (or have received this e-mail in error) 

please notify the sender immediately and destroy this e-mail. 
Any 
unauthorised copying, disclosure or distribution of the material in this

e-mail is strictly forbidden.
Thank you.

PR #131 (2020-05-04): Add FILETIME passthrough to notification decorator
datetime has inferior precision when compared to the original FILETIME
transmitted by ADS. Add the ability to have the FILETIME value passed
through in order to support the full resolution optionally if the
caller requests it.

Closes #116.
commit dbfaea6b (2020-05-04): Add FILETIME passthrough to notification decorator

datetime has inferior precision when compared to the original FILETIME
transmitted by ADS. Add the ability to have the FILETIME value passed
through in order to support the full resolution optionally if the
caller requests it.

Closes #116.