stlehmann/pyads #121: nuget package for testing? (conda-forge recipe)

klauer, 2020-04-17:
(I've recently been trying to help @stlehmann get his conda-forge recipe through to get pyads easily available on conda. Maybe this issue can be used as a side-band place to discuss a strategy to moving it forward without overwhelming the conda-forge reviewers.)

Getting to the point, Beckhoff has been supplying TwinCAT ADS libraries on the Windows package manager `nuget`: https://www.nuget.org/packages/Beckhoff.TwinCAT.Ads/4.4.4

I wonder if, for the testing phase, this could be installed on windows? I might just try a simple appveyor test to see if this will go through, as I have no Windows machine or VM without TwinCAT already installed.

klauer, 2020-04-17:
Nope, this was dumb - the nuget package only includes the .NET library, `TwinCAT.Ads.dll`, and not `TcAdsDll.dll`. I was hoping somehow the latter was a dependency of the .NET library...

closed 2020-04-17 by klauer

stlehmann, 2020-04-17:
Thanks for the try, though ;)