stlehmann/pyads #132: How to use Device Notifications in python class

ilya-vl, 2020-05-06:
Hello. Hi, is it possible to use Device Notifications in a class?
My example:

    class ADSserver(object):
        def __init__(self, parent = None):
            super(ADSserver, self).__init__()
            self.parent = parent

            AMS_id = "10.10.107.165.1.1"
            print("Connection to",AMS_id)
            self.plc = pyads.Connection(AMS_id, 851)
            self.plc.open()

            attr = pyads.NotificationAttrib(sizeof(pyads.PLCTYPE_UINT))
            handles = self.plc.add_device_notification('MAIN.iState', attr, self.callback)

        def callback(addr, notification, user_handle):
            print('Test')

        def test(self):
            pass

    if __name__ == "__main__":
        import time
        ads = ADSserver()
        time.sleep(1)`

This way it works, but I can't access another class function from the callback function.
How to pass the 'self' argument to the callback function?
Is it possible?

Thanks!

stlehmann, 2020-05-06:
Just add the self parameter to your callback method and you should be fine: 

    def callback(self, addr, notification, user_handle):
        print('Test')

Also consider to use the @notification decorator to make dealing with notifications easier.

ilya-vl, 2020-05-07:
I apologize for not immediately saying.
I tried adding the self argument.
```
    def callback(self, addr, notification, user_handle):
        self.test()
    def test(self):
        pass
```
But I get an error:
`TypeError: callback() missing 1 required positional argument: 'user_handle'`

___________________
As for the decorator:
I don't really need it yet, but I can't use it in class, either:

If I write as in the example:
```
  File "c:\Users\User\Desktop\ServoTracker\adsServer.py", line 20, in ADSserver
    @plc.notification(pyads.PLCTYPE_UINT)
NameError: name 'plc' is not defined
```
I get:
```
  File "c:\Users\User\Desktop\ServoTracker\adsServer.py", line 20, in ADSserver
    @plc.notification(pyads.PLCTYPE_UINT)
NameError: name 'plc' is not defined
```
And also, if I use a class object:
```
    @self.plc.notification(pyads.PLCTYPE_UINT)
    def callback(addr, notification, user_handle):
        pass
```
```
  File "c:\Users\User\Desktop\ServoTracker\adsServer.py", line 20, in ADSserver
    @self.plc.notification(pyads.PLCTYPE_UINT)
NameError: name 'self' is not defined
```
Thanks for the help!

ilya-vl, 2020-05-07:
ok, If you delete user_handle from the callback function, the function works.
The problem is resolved. Although I do not understand how :(

stlehmann, 2020-05-07:
That is strange. That would mean you only get addr and notification from the caller. Could you try to print self, addr and notification post it in here?

ilya-vl, 2020-05-07:
Yes, of course. But now my time is 00: 00. I can do this in about 9 hours. I will edit this message, or leave a new one so that you will receive a notification

ilya-vl, 2020-05-08:
notification

stlehmann, 2020-05-08:
That is not how it is supposed to be. Address and notification are swapped. Could you please post you system configuration: Opearating system, Python version, TwinCAT version.

ilya-vl, 2020-05-09:
TwinCAT 3.1.4024
Windows 10 home x64
python 3.8.2

stlehmann, 2020-05-11:
@ilya-vl I just tried your scenario on a Windows machine and was able to reproduce. I get the following Traceback:

```
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 237, in 'calling callback function'
  File "C:\Users\Lehmann\Documents\python\projects\pyads\pyads\pyads_ex.py", line 890, in wrapper
    return callback(notification, data)
TypeError: callback() missing 1 required positional argument: 'user_handle'
```

stlehmann, 2020-05-11:
The bug was probably introduced by #56. @lukabelingar can you remember why you added the wrapper clojure in pyads_ex.py:676 and why you omitted the user-handle argument? I know there was some good reason for it but can' t remember.

ilya-vl, 2020-05-11:
You need to add self as the first argument. Then there will be three

stlehmann, 2020-05-11:
This seems to be Windows-only.

stlehmann, 2020-05-11:
This should be fixed with #139. So I'm closing it now.

closed 2020-05-11 by stlehmann