stlehmann/pyads #133: Can pyads be used in macOS?

PaulNBarron, 2020-05-09:
Would there be any problems using pyads in macOS if I write a python script?

stlehmann, 2020-05-09:
There should be no problems using pyads on Mac OS.

PaulNBarron, 2020-05-09:
Great. I'm having problems making a connection to a Beckhoff PLC from macOS.
I have a mac and I'm running windows on a virtual machine using VM ware Fusion. On this I have TwinCAT and also run my python code here with pytest and it works fine. However I'd like to be able to run my python code without having to use the virtual machine but I'm having some trouble connecting.
The documentation says if you are using Linux you need to add a route one time, is it the same for macOS.  Does this still need to be done since I'm essentially using the same Ethernet connection?
If I try using the code in the quickstart guide I seem to get the error: 
pyads.pyads_ex.ADSError: ADSError: timeout elapsed (1861). 
2020-05-09T23:56:24+0200 Info: connection closed by remote
And if I try to use the add_route_to_plc() function I get the error:
OSError: [Errno 48] Address already in use

Any advice would be greatly appreciated.

stlehmann, 2020-05-10:
If you use the Connection class on Linux or Mac OS a route will be automatically created to the routing table of the client. However you still need to add a route to the routing table of the PLC for things to work. Do this either by adding it directly in you PLC or use the convenience function `add_route_to_plc`. 

The error you get seems like issue #66. This will be fixed with #134 which is currently in review. Please try setting the route manually until the fix is implemented.

stlehmann, 2020-05-10:
The bug should be fixed with #135. Please give it a try by using the master branch version directly from github.

stlehmann, 2020-05-13:
@PaulNBarron did #135 fix the issue for you?

stlehmann, 2020-05-15:
I'll close this for now.

closed 2020-05-15 by stlehmann

PaulNBarron, 2020-05-18:
No I haven't been able to get this to work yet. I had a lot of problems setting the route manually, the Beckhoff PLC just would disregard the details I entered when trying to add the route. I had to update the runtime version of TwinCAT and then it allowed me to successfully add a route on the PLC.
I've now gone back to MacOS and trying to get the add_route_to_plc to work but no success so far.

PaulNBarron, 2020-05-18:
Disregard, Sorry I didn't have the latest from the master branch installed. I've now managed to get it to work. Thanks a lot for the support!!

stlehmann, 2020-05-18:
> Links:

Glad to hear this works, now :)

Regards

---
Freundliche Grüße
Stefan Lehmann