stlehmann/pyads #145: Unable to connect

Prakyathkantharaju, 2020-07-15:
Hello,
I am trying to connect the twincat runtime which is running in the PC with the python to read and write data. Currently I am facing the following issue.
ERROR!!
```
Traceback (most recent call last):
  File ".\test.py", line 9, in <module>
    print(plc.read_by_name('VelocityRead', pyads.PLCTYPE_DINT))
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\ads.py", line 898, in read_by_name
    return adsSyncReadByNameEx(
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\pyads_ex.py", line 777, in adsSyncReadByNameEx
    handle = adsGetHandle(port, address, data_name)
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\pyads_ex.py", line 725, in adsGetHandle
    handle = adsSyncReadWriteReqEx2(
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\pyads_ex.py", line 600, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: target port not found   ADS Server not started (6).
```
I have setup the twincat router as follows. ![image_twincat](
[image]

ams id
[router_1.pdf](https://github.com/stlehmann/pyads/files/4922246/router_1.pdf)

Note that we are using the simulink for the motor control. 
[image]

Main Goal is to read/write the signal using python so that we can have a simple way to change the parameters. 

I the using the following the code:

```
import pyads

# add route to remote plc
# pyads.add_route("169.254.184.148.1.1", "169.254.234.136")

# connect to plc and open connection
print(pyads.PORT_SPS1)
plc = pyads.Connection('169.254.234.136.1.1', pyads.PORT_SPS3)
plc.open()

# read int value by name
i = plc.read_by_name("GVL.VelocityRead", pyads.PLCTYPE_DINT)
print(i)

# write int value by name
# plc.write_by_name("GVL.int_val", i, pyads.PLCTYPE_INT)

# close connection
plc.close()
```

ERROR!!
```
Traceback (most recent call last):
  File ".\test.py", line 9, in <module>
    print(plc.read_by_name('VelocityRead', pyads.PLCTYPE_DINT))
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\ads.py", line 898, in read_by_name
    return adsSyncReadByNameEx(
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\pyads_ex.py", line 777, in adsSyncReadByNameEx
    handle = adsGetHandle(port, address, data_name)
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\pyads_ex.py", line 725, in adsGetHandle
    handle = adsSyncReadWriteReqEx2(
  File "C:\Users\RRLAB-PK\AppData\Local\Programs\Python\Python38-32\lib\site-packages\pyads\pyads_ex.py", line 600, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: target port not found   ADS Server not started (6).
```

stlehmann, 2020-07-15:
Make sure you are using the right port for connection. In your example you are using PORT_SPS3 which is unlikely to work. You can investigate which port your plc is using in yourTwinCAT project (SPS stands for PLC):

[image]

Also make sure your TwinCAT system is in RUN mode.

Prakyathkantharaju, 2020-07-18:
@stlehmann Thank you for the reply!!
I am not using PLC or PLC project. its TwinCAT XAR installed on PC. for project we are using Simulink Model and its linked to A Task which has Port 350
[image]

I am trying to access variable: 'VelocityRead' from the TcCOM object 
[image]

We tried this code also changed the variable name to 'test.TcModuleInput.VelocityRead' but it gave the same error.

```
import pyads

# add route to remote plc
# pyads.add_route("169.254.234.136.1.1", "169.254.234.136")

# connect to plc and open connection
plc = pyads.Connection('169.254.234.136.1.1', 350)
plc.open()
print(plc.read_by_name('VelocityRead', pyads.PLCTYPE_DINT))
plc.close()
```
it gives the following error.
[image]

stlehmann, 2020-07-20:
As the error states the symbol can not be found. I'm not sure if it works but try to use the fully qualified name for your symbol: "test.TcModuleInput.VelocityRead". If this does not work either try to use a variable in the Global variables list as a proxy.

closed 2020-10-24 by stlehmann