stlehmann/pyads #146: Read structure with a PLCTYPE_ULINT Variable

LoicBoileau, 2020-07-16:
I have been trying to read a structure from my PLC through ADS. The following structure worked well : 
structDef = (('TimeStamp',pyads.PLCTYPE_INT,1),
			('CallerID', pyads.PLCTYPE_STRING,1,16),
			('bState', pyads.PLCTYPE_BOOL,1))

But when I switched for a PLCTYPE_ULINT for the 'TimeStamp' variable, the function doesnt work. I went to look in the pyads library and realised not every PLCTYPE are included in the structure/bytes conversion. (See picture)
[image]

I did proceed to add the PLCTYPE_ULINT to the dictionnary :
[image]
(I also included it in the file header)
[image]

Yet there still seems to be an issue related to the formatted bytes.
[image]

I hope you can fix this, meanwhile I will try myself.

Thanks,
Loic

PS. I did try to format my 'TimeStamp' variable with an PLCTYPE_LREAL before but it also seemed to not work.

stlehmann, 2020-07-19:
Seems like the ULINT datatype is not yet supported for structures. You already made some nice modifications. Would you like to create a PR for it so we can implement it in the master branch?

LoicBoileau, 2020-07-20:
Well, I would need to investigate the UnicodeDecodeError. For now, I don't think I have the time to do it but if I find some, it would be a nice improvement. Thanks for the quick answer! Hopes somebody else finds it useful !

stlehmann, 2020-07-20:
@LoicBoileau The format characters seem right to me. Maybe the structure definition on the plc side differs in you case.

LoicBoileau, 2020-08-04:
I don't think it differs since I did copy paste it on my python script, but I found a work-around: It seems that I have a problem with 64 bit data, so what i did is use a string for the TimeStamp Variable. Thanks for the answer!

closed 2020-10-24 by stlehmann