stlehmann/pyads #148: How to read an array that changes rapidly?

jishengzh3, 2020-07-28:
Hi Stefan,

Thank you very much for contributing to this awesome pyads lib.  

Recently, I've been trying to read 16 channels of simulated signals at 1kHz using a command like "_Data = plc.read_by_name('MAIN.Data',pyads.PLCTYPE_ARR_REAL(16),handle=hData)_", where _MAIN.Data_ is an array of real that stores 16 elements sampled at 1kHz. My purpose is to record the complete signal stream to display on GUI and meanwhile append to CSV file. However, my tests indicated the data wasn't recorded completely at 1kHz, where some time steps were skipped from pyads' reading. I could understand Python may not be able to capture the fast sampling. So, I'd like to ask for your advice on what is a  proper way to live-stream a complete measurement process? I wouldn't mind some delay in receiving on the Python client-side, but complete time steps are critical for my laboratory experimental measurements.

Your help would be much appreciated

Best regards,
Jisheng

stlehmann, 2020-07-28:
@jishengzh3 I'm glad you like the pyads package.

Realtime data is always critical. I only use pyads for samplerates of min. 100ms. If you experience problems with your higher samplerates a sensitive solution would be to aggregate multiple consecutive values in an array and send them cyclically via DeviceNotifications. In your case if you decide to buffer 10 consecutive values this would result in an array of size [16, 10]. Transferring that array at once is more effiient then to make 10 separate requests. Using the handle parameter also reduces request-time. But you already do that 👍 

I hope it helps a bit. Feel free to post your experiences here.
Regards
Stefan

jishengzh3, 2020-07-29:
@stlehmann Hi Stefan, Thank you very much for your prompt reply and great help. Following your tips, now I'm able to retrieve my data every 100ms. What I did was to set a periodic bool var in TC3, as briefly shown below. It worked for me now, but I'm not sure if the codes are correct. The other thing: I'm still a bit confused about how to cope with multiple notifications (if they could occur simultaneously) - do I need to create different threads for them separately? 

Cheers,
Jisheng

//-----------------------------------
// MAIN
bTrig : BOOL:=FALSE;
imod : ULINT := 100; //ms
nCount : ULINT;
//--------- code below ------------
nCount := nCount + 1;
IF (nCount MOD imod = 0)
THEN
    bTrig := NOT bTrig;
END_IF
//-----------------------------------

## Python client side:

import os
import pyads
import numpy as np
from ctypes import sizeof
import threading

plc = pyads.Connection('127.0.0.1.1.1', 851) 
plc.open()

def read_data(s = 'MAIN.Data'):   
    ## simulation 4 channels sampled at 1kHz
    hData = plc.get_handle(s)
    data = plc.read_by_name('MAIN.Data',pyads.PLCTYPE_ARR_REAL(400),handle=hData)
    data = np.reshape(data,(100,4))
    print(data)
    plc.release_handle(hData)
    return data

@plc.notification(pyads.PLCTYPE_BOOL)
def mycallback(hNotification,name,timestamp,value):
    pass

with plc:
     attr = pyads.NotificationAttrib(sizeof(pyads.PLCTYPE_BOOL),max_delay=0.0001)
     hNotification = plc.add_device_notification('MAIN.bTic', attr, mycallback)
     while True:
         data = read_data()
         pass

#plc.release_handle(hNotification)
#plc.close()

## End

stlehmann, 2020-07-30:
> @stlehmann [1] Hi Stefan, Thank you very much for your prompt reply
> Links:

Hello Jisheng,

would you mind continuing this conversation on Github so others can 
follow up.

Kind Regards

---
Freundliche Grüße
Stefan Lehmann

jishengzh3, 2020-07-31:
Hi Stefan,

Thank you very much for your help. Following your suggestions, I have resolved the issue by storing the samples into 2D arrays up to certain rows (e.g. 100 for each signal channel), and then sending them as packed a structure to the pyads client . The codes are shown below:

// PLC:
//-------------------------
// --- GVL
TYPE sample_structure:
STRUCT
	//bTrig : BOOL := FALSE;
	val1 :  ARRAY [1..100] OF REAL;
	Val2 : 	ARRAY [1..100] OF REAL;
END_STRUCT
END_TYPE
// --- MAIN
	n := n + 1;
	t := t + 0.001;
	val1[n] := t;
	val2[n] := SIN(t);
	IF (n MOD imod = 0) THEN
		st_vals.val1 :=val1;
		st_vals.val2 :=val2;
		n := 0;
	END_IF	
//-------------------------

### Client pyads
import pyads
from ctypes import sizeof

plc = pyads.Connection('127.0.0.1.1.1', 851) 
plc.open()
structure_def = (('val1',pyads.PLCTYPE_REAL,100),
                 ('val2',pyads.PLCTYPE_REAL,100))
size_of_struct = pyads.size_of_structure(structure_def)

@plc.notification(size_of_struct)
def callback(handle, name, timestamp, value):
    values = pyads.dict_from_bytes(value, structure_def)
    #print(f'Recived new notification for {name}.')
    print(values)
 
with plc:
    attr = pyads.NotificationAttrib(sizeof(size_of_struct))
    handles=plc.add_device_notification('MAIN.st_vals', attr, callback)
    while True:
        pass   

### END

stlehmann, 2020-07-31:
Glad this worked out for you. I assume the issue is resolved and will close it. Feel free to open it if you need further help.

closed 2020-07-31 by stlehmann