stlehmann/pyads #153: read string

amarco, 2020-08-09:
when I do a read_by_name and set the type to PLCTYPE_STRING i get back the first character, if I purposely tell it to read PLCTYPE_STRING *9999 something extremely I get back the characters in string plus a boat load of \x0.
I am wondering is there a method that can just return and arbitrary length string, or am I required to know the length?

Thanks

stlehmann, 2020-08-17:
You don't need to know the length of a string. Just pass PLCTYPE_STRING and you should be fine. For further questions I kindly ask you to add a minimal reproducable code sample that demonstrates your problem.

Kind Regards
Stefan

closed 2020-08-17 by stlehmann

amarco, 2020-08-17:
I have a string in twincat defined as
VAR_GLOBAL
                MagicID: WSTRING := "BRIDGERIG";
END_VAR

In my python code of I do

currentState=self.plc.read_by_name("GVL_ExcitAxis.MagicID", pyads.PLCTYPE_STRING)

currentState contains the letter ‘B’ only
Name
Value
Type
currentState
'B'
str

If I adjust python code to read 10 character
currentState=self.plc.read_by_name("GVL_ExcitAxis.MagicID", pyads.PLCTYPE_STRING *10 )

I get an array of letters where each letter is separated the \x0 for instance
Name
Value
Type
▶
currentState
[b'B', b'\x00', b'R', b'\x00', b'I', b'\x00', b'D', b'\x00', b'G', b'\x00']
list

So it appears the in one case it returns a list where every other item in list is the character and the other case it returns a type string of one character.

So my question still remains on how to read strings from a plc variable without know the length of the string?

Marco

RWDI - A Platinum Member of Canada's 50 Best Managed Companies
This communication is intended for the sole use of the party to whom it was addressed and may contain information that is privileged and/or confidential. Any other distribution, copying or disclosure is strictly prohibited. If you received this email in error, please notify us immediately by replying to this email and delete the message without retaining any hard or electronic copies of same. Outgoing emails are scanned for viruses, but no warranty is made to their absence in this email or attachments. If you require any information supplied by RWDI in a different format to facilitate accessibility, contact the sender of the email, email solutions@rwdi.com or call +1.519.823.1311.
Please be aware that when you contact us with a business query we may collect and use your details for future communications.

amarco, 2020-08-17:
Ok your comment made me think..

In Twincat there is a type STRING and a type WSTRING.
I was using twincat type WSTRING when I changed it to type STRING then python got back what I was expecting and it was of type string.
So looks like the library could use a PLCTYPE_WSTRING.

Marco