stlehmann/pyads #156: ADSError: target port not found   ADS Server not started (6).

samsara87, 2020-09-24:
OS = "Oracle Linux 7"
Version = 7.5

The application is running in a docker container.
Actually is the application not able to create a connection to the PLC.
Running the same code on Windows everything works fine.

For linux I have to use an additional line in the code 
pyads.set_local_address(local_net_id) // xxx.xxx.xxx.xxx.1.1

Communication from Server to PLC over port 48898 is working. Tested with telnet.

Any ideas what could be the reason for the port error?
Each kind of help appreciated.

stlehmann, 2020-09-24:
On Linux you need to add a route to your plc manually.

samsara87, 2020-09-24:
Already done. 
Route is added properly.
The way how I did.
- Added the route with the system manager (windows)
- Connected to the added PLC and I went to System and added the route manually.

stlehmann, 2020-09-24:
Can you please provide a minimal code example that describes your approach.

samsara87, 2020-09-24:
is that enough to understand the approach how I want to open the connection?

port=pyads.open_port()
print(port)
pyads.set_local_address(local_net_id)
pyads.close_port()

plc = pyads.Connection(ams_net_id, ads_port)#, ip_address=ip_address)
       print(ads_port)
       plc.open()
       yield plc

stlehmann, 2020-09-24:
Please also post the Traceback.

samsara87, 2020-09-24:
problem solved. Network issue. The configured network card (virtual) was moved to another host and got another mac address. So the hyper-v server was not working properly and was not able to access the plc through the firewall.

closed 2020-09-24 by samsara87