stlehmann/pyads #158: Struggle to run tests on Windows laptop

chrisbeardy, 2020-10-02:
I struggle to run the ADS test server on my windows laptop, I often get the follwing:

```
  File "C:\Users\chrisb\GitWS\pyads\tests\test_connection_class.py", line 39, in setUpClass
    cls.test_server = AdsTestServer(logging=True)
  File "c:\users\chrisb\gitws\pyads\.tox\py37\lib\site-packages\pyads\testserver.py", line 108, in __init__
    self.server.bind((self.ip_address, self.port))
OSError: [WinError 10013] An attempt was made to access a socket in a way forbidden by its access permissions
```

I also get the following for Python3.8

```
ERROR: test_ads (discover.ModuleImportFailure)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "c:\users\chrisb\gitws\pyads\.tox\py38\lib\site-packages\discover.py", line 64, in testFailure
    raise exception
ImportError: Failed to import test module: test_ads
Traceback (most recent call last):
  File "c:\users\chrisb\gitws\pyads\.tox\py38\lib\site-packages\discover.py", line 288, in _find_tests
    module = self._get_module_from_name(name)
  File "c:\users\chrisb\gitws\pyads\.tox\py38\lib\site-packages\discover.py", line 266, in _get_module_from_name
    __import__(name)
  File "C:\Users\chrisb\GitWS\pyads\tests\test_ads.py", line 9, in <module>
    import pyads
  File "c:\users\chrisb\gitws\pyads\.tox\py38\lib\site-packages\pyads\__init__.py", line 11, in <module>
    from .ads import (
  File "c:\users\chrisb\gitws\pyads\.tox\py38\lib\site-packages\pyads\ads.py", line 19, in <module>
    from .pyads_ex import (
  File "c:\users\chrisb\gitws\pyads\.tox\py38\lib\site-packages\pyads\pyads_ex.py", line 65, in <module>
    _adsDLL = ctypes.WinDLL("TcAdsDll.dll")  # type: ignore
  File "C:\Users\chrisb\AppData\Local\Programs\Python\Python38\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
```

Does someone have any insight to why this happens?

chrisjbremner, 2020-10-02:
I've run into the first error before, and it was because the Twincat processes were running in the background and interfering with the tests. Once I killed all the processes, I was able to run the tests.

chrisbeardy, 2020-10-03:
thanks, II've tried both deactivating twincat through the TcSwitchRuntime:

[image]

and also by stopping the service in the Services or just killing the System Service process in task manager, but when I do that I then get the following:

```
Traceback (most recent call last):
  File "C:\Users\chrisb\GitWS\pyads\tests\test_connection_class.py", line 228, in test_write_uint
    with self.plc:
  File "c:\users\chrisb\gitws\pyads\.tox\py37\lib\site-packages\pyads\ads.py", line 401, in __enter__
    self.open()
  File "c:\users\chrisb\gitws\pyads\.tox\py37\lib\site-packages\pyads\ads.py", line 413, in open
    self._port = adsPortOpenEx()
  File "c:\users\chrisb\gitws\pyads\.tox\py37\lib\site-packages\pyads\pyads_ex.py", line 303, in adsPortOpenEx
    raise RuntimeError("Failed to open port on AMS router.")
RuntimeError: Failed to open port on AMS router.
```

If i then try to restart the service, but do not put TwinCAT into config mode, it defaults to red TwinCAT Stopped:
[image]

I get the following error:

```
Traceback (most recent call last):
  File "C:\Users\chrisb\GitWS\pyads\tests\test_connection_class.py", line 233, in test_write_uint
    plc_datatype=constants.PLCTYPE_UDINT,
  File "c:\users\chrisb\gitws\pyads\.tox\py36\lib\site-packages\pyads\ads.py", line 510, in write
    self._port, self._adr, index_group, index_offset, value, plc_datatype
  File "c:\users\chrisb\gitws\pyads\.tox\py36\lib\site-packages\pyads\pyads_ex.py", line 514, in adsSyncWriteReqEx
    raise ADSError(error_code)
pyads.pyads_ex.ADSError: ADSError: Port disabled (18).
```

@chrisjbremner do you know what processes you had to kill? I wonder if I can somehow enable the port without having the service...

chrisbeardy, 2020-10-03:
I managed to solve the python38 issue by copying the appropriate (in my case the x64) `TcAdsDll.dll` file from `C:\TwinCAT\AdsApi\TcAdsDll\x64` to `C:\Windows\System32`. Not sure if this is caused by the way tox tests or my windows machine. But this just then rplicates the ADS Errors of 3.6 and3.7 now too.

chrisjbremner, 2020-10-03:
Now that you mention those additional errors, I think what I said was wrong. I remember that I saw those too and gave up and tested using the Docker container instead. Sorry for misleading you!

chrisbeardy, 2020-10-03:
No worries, I'll give the docker container a go to...

chrisbeardy, 2020-10-04:
I managed to get it working on docker, which will do for now, this does only to appear to be a problem with WIndows though.

stlehmann, 2020-10-24:
@chrisbeardy The tests do not run on Windows because the adslib does not work here. Testing only works on Unix machines or (as you did it) withing a virtual environment on Windows.

chrisbeardy, 2020-10-25:
closed, solved by #159

closed 2020-10-25 by chrisbeardy