stlehmann/pyads #163: Is pyads thread safe?

Adrian-at-CrimsonAuzre, 2020-11-04:
I'm having an issue where my program is crashing without throwing an exception. This program basically spawns a child thread to monitor a tag on a given machine and place the value in the database when it's changed. After a copious amount of print statements and debugging, I've found it only happens after attempting to read a tag off a Beckhoff PLC with this library.

Is there anything in pyads that isn't thread-safe? I'll be skimming through the code myself, but I figured I'd ask in the meantime.

stlehmann, 2020-11-04:
> This program basically spawns a child thread to monitor a tag on a given machine and place the value in the database when it's changed. 

That is actually an approach I use myself for GUI applications that use pyads so this should be fine.

> My guess is that the library is being called in two threads at the same time and something is happening.

There were some issues with the underlying adslib concerning DeviceNotifications with Async I/O. But they were fixed in https://github.com/stlehmann/ADS/commit/6684d9308dc79993924b91c62fac4941d2d68f69.

Adrian-at-CrimsonAuzre, 2020-11-04:
I'll update to the latest release and report back.

Adrian-at-CrimsonAuzre, 2020-11-04:
It doesn't seem to help unfortunately.

stlehmann, 2020-11-04:
@Adrian-at-CrimsonAuzre could you elaborate on the error a bit and maybe provide a minimal example that provokes the error. Thks 😄

Adrian-at-CrimsonAuzre, 2020-11-05:
The program just stops running, no error or anything. If I run it with trace `python -u -m trace -t PLC_watchdog.py` it doesn't crash. If I put all the pyads calls in a threading.Lock(), it also doesn't crash. This isn't ideal.
I'm guessing it's some sort of race condition within pyads or adslib.

stlehmann, 2020-11-05:
> The program just stops running, no error or anything

I got this already in your first post. But in order to get to the point of the problem we need you to provide a sample code that reproduces the described behaviour. Otherwise we won't be able to dig deeper into it. Ideally this code should be as minimalistic as possible. Can you do that?

Adrian-at-CrimsonAuzre, 2020-11-05:
Multiple of these threads are spawned by a parent thread, one for every tag of interest. 

```
class PLC_FetchThread(threading.Thread):
	def __init__(self, machine, *args, **kwargs):
		self.machine = machine
		super(PLC_FetchThread, self).__init__(*args, **kwargs)
		self._stop = threading.Event()

	def stop(self):
		self._stop.set()

	def stopped(self):
		return self._stop.isSet()

	def run(self):
		'''
		self.machine values:
		machine_id, machine_name, workcell_id,
		tag_id, tag_name,
		virtual_tag_id, virtual_tag_name, tag_poll_rate_ms,
		tag_type_id, type_name,
		plc_id, ip_address, ams_snap_address, username, password,
		plc_type_id, plc_type
		'''
		while not self.stopped():
			# Returns 0 when offline, 1 when running, and 2 when not connected
			# If we can't contact the self.machine, then return false because /something/ is offline
			try:
				# Pick out our Beckhoff datatype
				pyads_data_type = {'bool':pyads.PLCTYPE_BOOL, 'int':pyads.PLCTYPE_INT, 'string':pyads.PLCTYPE_STRING, 'float':pyads.PLCTYPE_REAL}.get(self.machine.type_name)
				# Connect to PLC and request tag data
				with pyads.Connection(self.machine.ams_snap_address, pyads.PORT_SPS1, self.machine.ip_address) as plc:
					tag_value = plc.read_by_name(self.machine.tag_name, pyads_data_type)

				### Sleep and database stuff
```

I think two of these threads are trying to read tags either from the same PLC, or from two different PLCs at the same time. Locking just the `plc.read_by_name` prevents it from crashing.

stlehmann, 2020-11-06:
I don't think it is a good idea to spawn multiple threads like this. You should have one thread that handles your ADS communication and another thread to handle your GUI for example.

Adrian-at-CrimsonAuzre, 2020-11-09:
So it turns out the issue isn't pyads or any of the other libraries I'm using. The issue seems to be with command prompt and bash.
On both Windows and Linux, after running for a random amount of time, the program just stops. However, if I daemonize the program or put it in a Docker container it will run continuously with no problems what so ever.

closed 2020-11-09 by Adrian-at-CrimsonAuzre