stlehmann/pyads #164: Get pyads.PLCTYPE from string?

RobertoRoos, 2020-11-05:
Is it possible to get a plc type based on a given string? I'm reading a model structure from an XML and connecting to a running TwinCAT instance.
E.g. I will read "LREAL" somewhere and want to map it to pyads.PLCTYPE_LREAL.

I could create a look-up dictionary for this, but I was hoping something like this already exists.

stlehmann, 2020-11-06:
A lookup-dict is definetely the way to go here.

RobertoRoos, 2020-11-07:
Alright. Ended up relying on the `hasattribute` function to omit a large list, as well as a separate lookup for array types:

```
class PlcTypeConverter:
    """Class with only static methods to handle PLC data-type identification

    This is a class instead of global functions so different parts and look-up dictionaries are neatly grouped together.

    Set the `strict` property to `True`
    """

    type_map_array = {
        'real': pyads.PLCTYPE_ARR_LREAL,  # LREAL, not REAL
        'int32': pyads.PLCTYPE_ARR_DINT,
        'int16': pyads.PLCTYPE_ARR_INT,
        # Some types have not been defined by pyads, we do it manually:
        'uint8': lambda n: pyads.PLCTYPE_USINT * n,
        'boolean': lambda n: pyads.PLCTYPE_BOOL * n,
    }

    strict = False

    def __init__(self):
        raise NotImplementedError('Class is only static!')

    @staticmethod
    def get_from_string(type_str: str):

        # If simple scalar
        plc_name = 'PLCTYPE_' + type_str
        if hasattr(pyads, plc_name):
            # Map e.g. 'LREAL' to 'pyads.PLCTYPE_LREAL' directly based on the name
            return getattr(pyads, plc_name)

        # If array/matrix (an 1D array is also called a matrix by Simulink)
        reg_match = re.match(r'matrix_(\d+)_(.*)_T', type_str)
        if reg_match is not None:

            groups = reg_match.groups()
            size = int(groups[0])
            scalar_type_str = groups[1]

            if scalar_type_str in PlcTypeConverter.type_map_array:
                return PlcTypeConverter.type_map_array[scalar_type_str](size)

        # The type could be a custom bus, which is pretty much impossible to map neatly
        # Set `strict` to `False` to quietly pass those type of variables

        if PlcTypeConverter.strict:
            raise NotImplementedError('The type `{0}` has not been mapped to a PLCTYPE'.format(type_str))
            # We can allow unmapped types at this point - Instead we throw an error when they are being addressed

        return None
```

closed 2020-11-07 by RobertoRoos