stlehmann/pyads #165: Get index_group and index_offset from PLC name

RobertoRoos, 2020-11-10:
Can I find the indx_gorup and index_offset based on the ADS variable name?
Basically I need to find a larger structure, and I think the easiest way would be using a variable name that I know will always be present.

Such a lookup feature could also be useful to (slightly) speed up repeated variable lookups (without getting and releasing handles).

RobertoRoos, 2020-11-14:
@stlehmann , this connects to [166] too. Any idea about this?

stlehmann, 2020-11-16:
I don't think we have an explicit function for this functionality. Do you know if this is supported by TwinCAT?

RobertoRoos, 2020-11-16:
There is. Beckhoff sample 14: https://infosys.beckhoff.com/english.php?content=../content/1033/tcsample_vc/html/TcAdsDll_API_CPP_Sample14.htm&id=

And it's actually already implemented in pyads too: `pyads.ex.adsGetSymbolInfo(...)`

I'm using this in the AdsSymbol extension.
We can close this issue then.

closed 2020-11-16 by RobertoRoos