stlehmann/pyads #166: ADS Symbol class

RobertoRoos, 2020-11-10:
Let me start that I am loving this library! This is allowing my team and me to easily create an application around a TwinCAT solution in good 'ol familiar Python.

For a part of our application we created a Symbols class that wraps around an ADS variable. The goal was to reduce:

```python
index_group, index_offset = ...
plc.read(index_group, index_offset, pyads.PLCTYPE_LREAL)
plc.write(index_group, index_offset, 420.0, pyads.PLCTYPE_LREAL)
```

...to:

```
symbol = Symbol(plc, ...)
symbol.get()
symbol.set(420.0)
```

The Symbol class keeps a reference to the Connection and remembers the indices. (The example is a little trivial, but for the large number of variables in a tree structure this became very helpful.)

Would it be worthwhile to make such a class part of pyads itself?

RobertoRoos, 2020-11-11:
My bad, I only just found that `pyads.structs.AdsSymbol` already exists. Still, should be expand its functionality?

stlehmann, 2020-11-13:
@RobertoRoos I'm glad you like the package. I didn't think it through but having a symbol wrapper class as described by you seems quite charming to me. However, I would go more for read/write instead of get/set because we already use these terms
in the package. I could imagine a use-case like this:

```python
symbol = Symbol(plc, "global.plc_variable", PLCTYPE_INT)
value = symbol.read()
symbol.write(123)
``` 

Maybe even this which would be more Pythonic:

```python
>>> symbol = Symbol(plc, "globa.symbol", PLCTYPE_INT)
>>> value = symbol.value
>>> symbol.value = 123
>>> print(symbol.type)
PLCTYPE_INT
>>> print(symbol.index_group)
...
>>> print(symbol.index_offset)
...
```

tboegi, 2020-11-16:
You don't need to specify the type like this:
symbol = Symbol(plc, "globa.symbol", PLCTYPE_INT)

That one should be enough:
symbol = Symbol(plc, "global.symol")

There is some example code here, which may be helpful:
https://infosys.beckhoff.com/english.php?content=../content/1033/tcsample_vc/html/tcadsdll_api_cpp_sampleintro.htm&id=

stlehmann, 2020-11-16:
@tboegi Which example are you referring to exactly?

tboegi, 2020-11-16:
Sorry, the example-number did not went into my copy-paste.
The trick is to use index group 0XF009:

ADSIGRP_SYM_INFOBYNAMEEX 16#F009

I think it is exmple 14, as pointed out in #165 
It seams somewhat "incomplete" (this are my words), I can dig out some
code that uses F009, but that is happening tomorrow.

bdcoyle, 2020-11-16:
Perhaps most of the functionality discussed here is already covered by the 'read_list_by_name' and 'write_list_by_name' functions that were recently added.

https://pyads.readthedocs.io/en/latest/pyads.html#pyads.ads.Connection.read_list_by_name
https://pyads.readthedocs.io/en/latest/pyads.html#pyads.ads.Connection.write_list_by_name

read_list_by_name takes a list of variable names as argument. It then reads the symbol info for each variable using the index group referred to by @tboegi. The symbol info is cached in the Connection class by default so this lookup only needs to be done on the first use of each variable. 
It then uses the index group and index offset of each variable to do a sum read of the variables, which means the values of all variables are read in a single transaction with the PLC.

write_list_by_name works in the same way, except it takes a dict as argument, with the variable name and value as key-value pair.

tboegi, 2020-11-17:
Sorry for my "oldish" comments - digging into the latest pyads sources shows,
that we already have everything here - since Octover 2020 (?).
All the stuff is (already) in SAdsSymbolEntry. 
Nice :-)

stlehmann, 2020-11-17:
I think the proposal of @RobertoRoos is to expand the functionality of the "dumb" SAdsSymbol class (which is basically a container by now) to a fully functional interface for reading and writing data to the connected ADS device. It seems like a promising idea for me.

**Pros:**
- lighter code
- one can create all the symbols and pass them along the application so no need for a lookup
- more pythonic

**Cons:**
We need to pass a reference to the Connection to AdsSymbol.
There will be multiple AdsSymbol objects in one application which refer to one Connection object.
So if this connection is closed or destroyed the AdsSymbol objects need to handle this either by
re-opening the connection or by throwing an exception.

I could imaging going about it this way:
1. add a value property that stores the current value
2. add a read method that returns the value from the plc and sets self.value
3. add a write method with a value paramter that either writes value to the plc or if value is None write self.value to the plc

So basically I could think of an implementation like this:

```python
class AdsSymbol:
    def __init__(self, connection, index_group, index_offset, name, symtype, comment):
        ...
        self.connection = connection
        self.index_group = index_group
        self.index_offset = index_offset
        self.name = name
        self.symtype = symtype
        self.comment = comment
        self.value = None

    def read(self):
         value = self.connection.read(self.index_group, self.index_offset, ...)
         self.value = value
         return value

    def write(value=None):
        if value is None:
            value = self.value

        self.connection.write(self.index_group, self.index_offset, ...)
        self.value = value
```

RobertoRoos, 2020-12-11:
@stlehmann I just placed a pull request for this.
First contribution, so thank you for your patience in getting it right!

PR #174 (2021-01-10): Add AdsSymbol class
Expanded the `AdsSymbol` class to stand on its own. Also updated the `AdvancedHandler` tester class.
Fixes #166 and related to #168
closed 2021-01-10 by stlehmann