stlehmann/pyads #168: Problems with getting symbol info from AdsTestServer

RobertoRoos, 2020-11-17:
I am writing the tests for #166 and having some trouble with getting it to work with the test server.

I copied the `AdsConnectionClassTestCase` test class and named it `AdsSymbolTestCase`.

Test code like this:

```
        with self.plc:
            symbol = AdsSymbol(self.plc, handle_name)
            print('Symbol:', symbol)
            print('Value:', symbol.value)
```

....fails because the type could not be determined using `pyads_ex.adsGetSymbolInfo`. The `type_name` field of the info struct is an empty string.

How can we populate the `AdsTestServer` with test variables?  
How does `AdsConnectionClassTestCase.test_get_all_symbols_single` create the variable that is being returned?

stlehmann, 2020-12-10:
@RobertoRoos is this still an issue? I just came across this issue but couldn't find the tests you were referring to.

RobertoRoos, 2020-12-10:
Yeah, haven't made much progress.

I see haven't pushed any tests yet, but haven't written anything useful yet.

It think it's best to ask it as an open question: How can I make `adsGetSymbolInfo()` return a PLCTYPE from the testserver?
The `BasicHandler` always returns 0, which does not correspond to an actual.

RobertoRoos, 2020-12-10:
Making some progress. Basically, there was no support for a `ADSIGRP_SYM_INFOBYNAMEEX` type request in the `AdvancedHandler`. Currently writing this.

And to answer my original question: `AdvancedHandler` remembers the variables addressed in previous requests. So you can create remote variables simply writing something to it.

stlehmann, 2021-01-10:
@RobertoRoos Just to clean it up: Has this issue been resolved with your latest PR?

RobertoRoos, 2021-01-10:
Yes, the current `AdsSymbolTestCase` gives an example of how this works.

closed 2021-01-10 by RobertoRoos