stlehmann/pyads #170: pyads and mac Big Sur

Swaab, 2020-11-27:
Hi there! 

I got the following stack trace running pyads on 
- mac big sur 
- python 3.7/3.8/3.9

I tested it also on catalina, there it works
- mac catalina 
- python 3.8

Im not familiar with the inner workings and cant fix it sadly, is this something that can be fixed?  

  File "/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(/Users/../.local/share/virtualenvs/WebBackend--zKZU3uM/lib/python3.8/site-packages/pyads/adslib.so, 6): no suitable image found.  Did find:
	/Users/../.local/share/virtualenvs/WebBackend--zKZU3uM/lib/python3.8/site-packages/pyads/adslib.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00
	/Users/../.local/share/virtualenvs/WebBackend--zKZU3uM/lib/python3.8/site-packages/pyads/adslib.so: unknown file type, first eight bytes: 0x7F 0x45 0x4C 0x46 0x02 0x01 0x01 0x00

stlehmann, 2020-12-02:
@Swaab seems like the there was some error compiling adslib. Can you try to compile adslib manually?

stlehmann, 2020-12-17:
I will close this issue for now. Feel free to attach to it if the problem still persists.

closed 2020-12-17 by stlehmann

Swaab, 2021-01-22:
HI! srry was gone for a bit; Thx for your response!

I've been searching for a bit but don't quite understand how I can compile the adslib manually. Would you mind giving me some pointers?  Thx in advance!

stlehmann, 2021-01-24:
@Swaab You need to change in the directory adslib and run make.

Swaab, 2021-01-28:
Aaah I finally got it together; I was looking in the pyads folder within my venv and only say the adslib.so file and didn't really understand the structure; Apperntly the adslib folder get's installed higher in the venv hierarchy. I dit 'make' and copied the adslib.so file to the pyads folder; This fixet it; 

Making did give a warning, ill print it here for your reference: 
`mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AdsDef.cpp -o obj/AdsDef.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AdsDevice.cpp -o obj/AdsDevice.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AdsLib.cpp -o obj/AdsLib.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AdsNotification.cpp -o obj/AdsNotification.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AmsConnection.cpp -o obj/AmsConnection.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AmsPort.cpp -o obj/AmsPort.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/AmsRouter.cpp -o obj/AmsRouter.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/Frame.cpp -o obj/Frame.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/Log.cpp -o obj/Log.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/NotificationDispatcher.cpp -o obj/NotificationDispatcher.o -I AdsLib/
mkdir -p obj/
c++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/Sockets.cpp -o obj/Sockets.o -I AdsLib/
AdsLib/Sockets.cpp:196:69: warning: adding 'int' to a string does not append to the
      string [-Wstring-plus-int]
        throw std::runtime_error("Connect TCP socket failed with: " + WSAGetLastError());
                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~
AdsLib/Sockets.cpp:196:69: note: use array indexing to silence this warning
  ...std::runtime_error("Connect TCP socket failed with: " + WSAGetLastError());
                                                           ^
                        &                                  [                  ]
1 warning generated.
c++ -shared -fPIC -o adslib.so obj/AdsDef.o obj/AdsDevice.o obj/AdsLib.o obj/AdsNotification.o obj/AmsConnection.o obj/AmsPort.o obj/AmsRouter.o obj/Frame.o obj/Log.o obj/NotificationDispatcher.o obj/Sockets.o`