stlehmann/pyads #173: Duplicate classes for TestServer and Handlers

RobertoRoos, 2020-12-10:
Just noticed there is a class `Testserver` defined in 'pyads/testserver_ex/server.py' and a class `AdsTestServer` in 'pyads/testserver.py'.  
Similarly, there is a class `AdvancedHandler` in 'pyads/testserver.py' and a class `AdvancedHandler` in 'pyads/testserver_ex/handler.py'.

Is there a reason for these duplicates? It's a little confusing.

stlehmann, 2020-12-14:
Yes testserver_ex contains the prototype of an advanced testserver that can store values and returns them on write / read operations. I never managed to finish this prototype so it is still not used in any tests.

RobertoRoos, 2020-12-14:
Ah, right. I thought that maybe it was the opposite: a left over from older versions.

Note that I patched the existing AdvancedHandler in https://github.com/stlehmann/pyads/pull/174, maybe some of it could be helpful.

stlehmann, 2021-01-24:
@RobertoRoos Which classes do you mean pricesely?

RobertoRoos, 2021-01-24:
There are [testserver.AdsTestServer](https://github.com/stlehmann/pyads/blob/master/pyads/testserver.py#L79) and [testserver_ex.server.Testserver](https://github.com/stlehmann/pyads/blob/master/pyads/testserver_ex/server.py#L36)

And [testserver.AdvancedHandler](https://github.com/stlehmann/pyads/blob/master/pyads/testserver.py#L624) and [testserver_ex.handler.AdvancedHandler](https://github.com/stlehmann/pyads/blob/master/pyads/testserver_ex/handler.py#L35)

Just to emphasize: I am not at all against improving the testserver or switching to a new approach, but I don't think an unfinished or unused feature belongs in the production branch.

stlehmann, 2021-01-24:
I'm with you in this matter. If I remember right I created the testserver_ex package to bring more order in the testerver module. So the overall goal would be to replace the simple testserver module by this package. Maybe a way to go about would be to put testserver_ex in a new branch called "testserver_ex" to have a clean master branch as you suggested. What do you think?

RobertoRoos, 2021-01-24:
Yeah, let's do that.

stlehmann, 2021-02-11:
Thanks @RobertoRoos for getting this done.

PR #193 (2021-02-11): Removed developing test-server
Fixes #173 

I don't think I can create a new branch on this repo, so @stlehmann you'll need to create a branch at the current tip.
closed 2021-02-11 by stlehmann