stlehmann/pyads #176: Problems after update: struct.error: unpack requires a buffer of 1 bytes

Quack888, 2021-01-03:
Hello, 

i have a Twincat 2 Projekt runing at a CX1020 with normal boolean, real and other Global Variables.
I've written a code with pyads to visible the Values in my smarthomeng home automaition with python 3.5 some years ago.
The last to days i updatet my home automation and also upgrade my python version to 3.8 and here my issues starts:
Now i can't get an device notification running. Not for an boolena and also not for an real. So i tried the examples from the doc but i still not working. Here an example i tried in the python console:

>>> plc.open()
2021-01-03T22:50:53+0100 Info: Connected to 192.168.0.11
>>>
>>> @plc.notification(pyads.PLCTYPE_BOOL)
... def callback(handle, name, timestamp, value):
...     print('{}: recieved notification for {} :: {}'.format(timestamp, name, value))
...
>>>
>>> plc.add_device_notification('.Qx_HeizungspumpeEG', pyads.NotificationAttrib(2), callback)
(7965, 889192894)
>>> Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 237, in 'calling callback function'
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/pyads_ex.py", line 1097, in wrapper
    return callback(notification, data)
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 1088, in func_wrapper
    hNotification, timestamp, value = self.parse_notification(
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 1173, in parse_notification
    value = struct.unpack(DATATYPE_MAP[plc_datatype], bytearray(data))[0]
struct.error: unpack requires a buffer of 1 bytes
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 237, in 'calling callback function'
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/pyads_ex.py", line 1097, in wrapper
    return callback(notification, data)
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 1088, in func_wrapper
    hNotification, timestamp, value = self.parse_notification(
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 1173, in parse_notification
    value = struct.unpack(DATATYPE_MAP[plc_datatype], bytearray(data))[0]
struct.error: unpack requires a buffer of 1 bytes
Traceback (most recent call last):
  File "_ctypes/callbacks.c", line 237, in 'calling callback function'
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/pyads_ex.py", line 1097, in wrapper
    return callback(notification, data)
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 1088, in func_wrapper
    hNotification, timestamp, value = self.parse_notification(
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 1173, in parse_notification
    value = struct.unpack(DATATYPE_MAP[plc_datatype], bytearray(data))[0]
struct.error: unpack requires a buffer of 1 bytes

Then i tried to do the same by reading by list. But it also is not running:

>>> plc.open()
2021-01-03T22:47:39+0100 Info: Connected to 192.168.0.11
>>> var_list = ['.Qx_HeizungspumpeEG','.Qx_HeizungspumpeOG','.Ix_BrennerLauft']
>>>
>>> plc.read_by_name(var_list[0], pyads.PLCTYPE_BOOL)
True
>>> plc.read_by_name(var_list[1], pyads.PLCTYPE_BOOL)
True
>>> plc.read_by_name(var_list[2], pyads.PLCTYPE_BOOL)
False
>>>
>>>
>>> plc.read_list_by_name(var_list)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/ads.py", line 798, in read_list_by_name
    return adsSumRead(self._port, self._adr, data_names, data_symbols)
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/pyads_ex.py", line 830, in adsSumRead
    sum_response = adsSyncReadWriteReqEx2(
  File "/home/smarthome/.local/lib/python3.8/site-packages/pyads/pyads_ex.py", line 638, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: parameter size not correct (1797).
>>>

So can anyone help me with my problem or know a solution for me?

And sorry for my english!

RobertoRoos, 2021-01-05:
I think that in the first example you do not set the correct variable size in the notification attribute. To go from your example:

```
@plc.notification(pyads.PLCTYPE_BOOL)
... def callback(handle, name, timestamp, value):
... print('{}: recieved notification for {} :: {}'.format(timestamp, name, value))
...

attrib = pyads.NotificationAttrib(sizeof(pyads.PLCTYPE_BOOL))
plc.add_device_notification('.Qx_HeizungspumpeEG', attrib, callback)
```

So you'll need to specify your type twice: once in the notification set-up and once in the notification callback.

Quack888, 2021-01-05:
Thank you very much, your right. Now i had fixed my code and it is still running.

closed 2021-01-05 by Quack888