stlehmann/pyads #177: pyads.pyads_ex.ADSError: ADSError: target port not found   ADS Server not started (6).

christojet, 2021-01-04:
**This is my code:**
import pyads
# connect to plc and open connection
plc = pyads.Connection('192.168.10.1.1.1', pyads.PORT_SPS1)
plc.open()
# read int value by name
L1 = plc.read_by_name("POUs.LBK1Distance", pyads.PLCTYPE_INT)
print(L1)
# close connection
plc.close()

**This is what error i got:**
Traceback (most recent call last):
  File "pyadsTest.py", line 17, in <module>
    L1 = plc.read_by_name('LBK1Distance', pyads.PLCTYPE_INT)
  File "C:\Users\c.bousquet-jette\AppData\Local\Continuum\anaconda3\envs\ROCAP\lib\site-packages\pyads\ads.py", line 761, in read_by_name
    return adsSyncReadByNameEx(
  File "C:\Users\c.bousquet-jette\AppData\Local\Continuum\anaconda3\envs\ROCAP\lib\site-packages\pyads\pyads_ex.py", line 977, in adsSyncReadByNameEx
    handle = adsGetHandle(port, address, data_name)
  File "C:\Users\c.bousquet-jette\AppData\Local\Continuum\anaconda3\envs\ROCAP\lib\site-packages\pyads\pyads_ex.py", line 770, in adsGetHandle
    handle = adsSyncReadWriteReqEx2(
  File "C:\Users\c.bousquet-jette\AppData\Local\Continuum\anaconda3\envs\ROCAP\lib\site-packages\pyads\pyads_ex.py", line 638, in adsSyncReadWriteReqEx2
    raise ADSError(err_code)
pyads.pyads_ex.ADSError: ADSError: target port not found   ADS Server not started (6).

[image]

What can i do?

RobertoRoos, 2021-01-05:
The error is pretty descriptive: it seems you have the wrong port.

In Visual Studio, to the TwinCAT dropdown and select 'Target Browser' -> 'Target Browser'. In the Ads tab, expand the nodes to find the variable you're looking for. Note the port number of the top node, this is the port you will need to connect.

The value you are looking for is part of a struct? Maybe you could add a simple integer variable next to it and try to read it, just to make sure you can reach the PLC instance.

stlehmann, 2021-01-06:
@christojet Thanks for reaaching out. We recommend using Stackoverflow for general questions on using pyads and ADS in general. Github issues is for bug reporting and software development so I will close this issue.

closed 2021-01-06 by stlehmann

christojet, 2021-01-06:
Hi RobertoRoos,
When i go to the destination you mention, i see that:
[image]
Thanks you :)