stlehmann/pyads #183: Support structured ADS types with read_list_by_name

pylipp, 2021-01-12:
In a [different PR](https://github.com/stlehmann/pyads/pull/161#issuecomment-720324694) I wondered whether `Connection.read_list_by_name` is compatible with structured ADS variables? E.g.

```
>>> structure_def = (
...    ('rVar', pyads.PLCTYPE_LREAL, 1),
...    ('rVar2', pyads.PLCTYPE_REAL, 1),
...    ('iVar', pyads.PLCTYPE_INT, 1),
...    ('iVar2', pyads.PLCTYPE_DINT, 3),
...    ('sVar', pyads.PLCTYPE_STRING, 1)
... )
>>> plc.read_list_by_name(['global.sample_structure'])
```

@bdcoyle then said it's not, so I'd like to know whether I can assist here.

pylipp, 2021-01-13:
I think [this](https://infosys.beckhoff.de/english.php?content=../content/1033/tcsample_vc/html/TcAdsDll_API_CPP_Sample17.htm&id=) is a relevant resource but I cannot make out whether sum-read is compatible with structured datatypes.

stlehmann, 2021-01-13:
I don't think sum read/write is supported for structurer datatypes at the moment. Am I right @bdcoyle?

stlehmann, 2021-01-13:
I don't know if it is even supported by ADS itself.

bdcoyle, 2021-01-13:
That's correct - at the moment it's not supported.

I _think_ it should be possible - I assume a structure has an index group and index offset like any other variable in the PLC, these plus the structure size is what is needed to make the sum read/write requests.

A good point to start would be to call adsGetSymbolInfo on a structure in the PLC, and make sure the idx grp and idx offset are valid.

Then within sumRead and sumWrite, an extra case would need to be added to handle structures - it would need to unpack from the binary data returned by the PLC into the structure_def.
Maybe some of the existing functions for handling structures could be used here.

I don't have time at the moment to contribute on this, so feel free to give it a shot!

stlehmann, 2021-02-18:
Thanks to @pylipp we now also have this feature. I think I can close this issue as it is solved by 
#187

closed 2021-02-18 by stlehmann