stlehmann/pyads #184: Building documentation on readthedocs.org for package using pyads

RobertoRoos, 2021-01-12:
I am trying to get the docs of my package (which uses pyads of course) to be built and hosted by readthedocs.org. However, it runs into the following error:

> OSError: adslib.so: cannot open shared object file: No such file or directory

...which doesn't make the build fail, but all autodocs end up empty.

@stlehmann How did you get https://pyads.readthedocs.io/en/latest/ to build properly?

stlehmann, 2021-01-12:
@RobertoRoos make sure the pyads package is installed properly. The install process should include the compilation of adslib.so.
Make sure you enabled "Install Project" in the extended settings.

RobertoRoos, 2021-01-12:
I now checked the 'Install Project' setting, but it made no difference. I think this only influences how my package is installed, not pyads as a dependency.

I'll try running this in their Docker image tomorrow, see what happens.

Any other input would be greatly appreciated!

RobertoRoos, 2021-01-13:
Did some tests from an Ubuntu virtual machine

- Running `pip3 install pyads` (without this repo)  _is not enough_, `import pyads` will fail because of the missing library
    - This can be fixed by building and installing `pyads/adslib`
- Running `python3 setup.py install` from this pyads repository does work, adslib will be build and installed inside the package directory

(Apologies if this was obvious already)

Then it makes sense my build is failing. I need to make sure the pyads install is complete.
I guess one way would be to include this pyads repo as a submodule in my project, and have it installed using setuptools. However, this is kind of unfortunate since this is only to facility a docs build.

RobertoRoos, 2021-01-13:
Fixed it. Their FAQ tackles this exactly: https://docs.readthedocs.io/en/stable/faq.html#i-get-import-errors-on-libraries-that-depend-on-c-modules (silly that I did not find it before).

It's as simple as adding this line to my docs `conf.py`:

```
autodoc_mock_imports = ["pyads"]
```

As the setting suggests, this mocks the import of pyads.

Full repo: https://bitbucket.org/ctw-bw/twinpy (including `.readthedocs.yml`, my RTD project has no extra settings)

closed 2021-01-13 by RobertoRoos

stlehmann, 2021-01-13:
> Running pip3 install pyads (without this repo) is not enough, import pyads will fail because of the missing library

This comes to my utter surprise. I always assumed `pip install pyads` also compiles adslib.so Also I'm not aware of any issues concerning missing adslib.so.
However I just tried this on Ubuntu myself and made the same experience as you. adslib.so has not been compiled. I wonder if this is connected to the version of pip? Which version of Python / pip are you using?

stlehmann, 2021-01-13:
We will address this issue in #185.