stlehmann/pyads #185: Bug: pip install pyads does not compile adslib.so

stlehmann, 2021-01-13:
Firstly reported in issue #184 the command `pip install pyads` does not build the adslib.so on Unix systems as intended.

stlehmann, 2021-01-13:
Strangely if I create a local distribution via `python setup.py sdist` and install it via `pip install ./dist/pyadsxxxxx.tar.gz` the build process works like a charm.

RobertoRoos, 2021-01-13:
Problem can be reproduced with this `Dockerfile`:

```
FROM python:3.8

COPY script.py ./

RUN pip install --upgrade pip && \
    pip install pyads==3.3.1

RUN python ./script.py
```

```
# script.py
import pyads
print(pyads.Connection)
```

Docker build output:

```
$ sudo docker build -t pyads .
Sending build context to Docker daemon  3.072kB
Step 1/4 : FROM python:3.8
 ---> b0358f6298cd
Step 2/4 : COPY script.py ./
 ---> Using cache
 ---> 1a11a750be02
Step 3/4 : RUN pip install --upgrade pip &&     pip install pyads==3.3.1
 ---> Running in 801c877a90a5
Requirement already satisfied: pip in /usr/local/lib/python3.8/site-packages (20.3.3)
Collecting pyads==3.3.1
  Downloading pyads-3.3.1.tar.gz (44 kB)
Building wheels for collected packages: pyads
  Building wheel for pyads (setup.py): started
  Building wheel for pyads (setup.py): finished with status 'done'
  Created wheel for pyads: filename=pyads-3.3.1-py3-none-any.whl size=49109 sha256=6f7b9e6b5c6db997a222a2bf31adf0e915f89e2ade075e1f44457b1602948360
  Stored in directory: /root/.cache/pip/wheels/93/c0/30/bf8035cab649f97a54081395d90bbe93a0181d889efbe1f19e
Successfully built pyads
Installing collected packages: pyads
Successfully installed pyads-3.3.1
Removing intermediate container 801c877a90a5
 ---> f9dd90ec44b7
Step 4/4 : RUN python ./script.py
 ---> Running in 191a9731894a
Traceback (most recent call last):
  File "./script.py", line 1, in <module>
    import pyads
  File "/usr/local/lib/python3.8/site-packages/pyads/__init__.py", line 11, in <module>
    from .ads import (
  File "/usr/local/lib/python3.8/site-packages/pyads/ads.py", line 28, in <module>
    from .symbol import AdsSymbol
  File "/usr/local/lib/python3.8/site-packages/pyads/symbol.py", line 11, in <module>
    from .pyads_ex import adsGetSymbolInfo
  File "/usr/local/lib/python3.8/site-packages/pyads/pyads_ex.py", line 96, in <module>
    _adsDLL = ctypes.CDLL(adslib)
  File "/usr/local/lib/python3.8/ctypes/__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: adslib.so: cannot open shared object file: No such file or directory
The command '/bin/sh -c python ./script.py' returned a non-zero code: 1
```

RobertoRoos, 2021-01-13:
Spamming more logs. `pip install pyads==3.3.0 --verbose` gives some more info. For 3.3.0 we get:

>  Added pyads==3.3.0 from https://files.pythonhosted.org/packages/42/60/293dd529d25b93d77315396756abd85912d81d14f17663ae5117cf962240/pyads-3.3.0.tar.gz#sha256=8369069a33766b6488cda5aae5b3768b48ea5f3b9d4b43d8a2b6802e4b37ff70 to build tracker '/tmp/pip-req-tracker-1fzbh2as'
>     Running setup.py (path:/tmp/pip-install-4kik2zu5/pyads_5d5d6eb392c047f58fb90ab6d4cde274/setup.py) egg_info for package pyads
>     Created temporary directory: /tmp/pip-pip-egg-info-9edn5h70
>     Running command python setup.py egg_info
>     running egg_info                                                                                               
>     creating /tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info                                                         
>     writing /tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info/PKG-INFO                                                 
>     writing dependency_links to /tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info/dependency_links.txt                 
>     writing top-level names to /tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info/top_level.txt                         
>     writing manifest file '/tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info/SOURCES.txt'                              
>     reading manifest file '/tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info/SOURCES.txt'                              
>     writing manifest file '/tmp/pip-pip-egg-info-9edn5h70/pyads.egg-info/SOURCES.txt'                              
>   Source in /tmp/pip-install-4kik2zu5/pyads_5d5d6eb392c047f58fb90ab6d4cde274 has version 3.3.0, which satisfies requirement pyads==3.3.0 from https://files.pythonhosted.org/packages/42/60/293dd529d25b93d77315396756abd85912d81d14f17663ae5117cf962240/pyads-3.3.0.tar.gz#sha256=8369069a33766b6488cda5aae5b3768b48ea5f3b9d4b43d8a2b6802e4b37ff70
>   Removed pyads==3.3.0 from https://files.pythonhosted.org/packages/42/60/293dd529d25b93d77315396756abd85912d81d14f17663ae5117cf962240/pyads-3.3.0.tar.gz#sha256=8369069a33766b6488cda5aae5b3768b48ea5f3b9d4b43d8a2b6802e4b37ff70 from build tracker '/tmp/pip-req-tracker-1fzbh2as'
> Created temporary directory: /tmp/pip-unpack-k_wz6jkf
> Building wheels for collected packages: pyads
>   Created temporary directory: /tmp/pip-wheel-8fqgll1g
>   Building wheel for pyads (setup.py): started
>   Destination directory: /tmp/pip-wheel-8fqgll1g
>   Running command /usr/local/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-4kik2zu5/pyads_5d5d6eb392c047f58fb90ab6d4cde274/setup.py'"'"'; __file__='"'"'/tmp/pip-install-4kik2zu5/pyads_5d5d6eb392c047f58fb90ab6d4cde274/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-8fqgll1g                                                                                                       
>   running bdist_wheel                                                                                              
>   running build                                                                                                    
>   make: Entering directory '/tmp/pip-install-4kik2zu5/pyads_5d5d6eb392c047f58fb90ab6d4cde274/adslib'               
>   mkdir -p obj/                                                                                                    
>   g++ -c -std=c++11 -pedantic -Wall -fPIC AdsLib/Sockets.cpp -o obj/Sockets.o -I AdsLib/                           
>   mkdir -p obj/
>   ...

And with 3.3.1:

>  Added pyads==3.3.1 from https://files.pythonhosted.org/packages/58/d7/3194cfc564444685bdcbf94434e3c2ac82424894a9fae8a17ffe4cb9b665/pyads-3.3.1.tar.gz#sha256=e0d010c620e3e5a47c36035d7d43cdf3d1e9cc9a27c99de128485690b7824508 to build tracker '/tmp/pip-req-tracker-awepnpo4'
>     Running setup.py (path:/tmp/pip-install-xl4avn05/pyads_79d9cc3840c64db4b21093943835add6/setup.py) egg_info for package pyads
>     Created temporary directory: /tmp/pip-pip-egg-info-3f15zffe
>     Running command python setup.py egg_info
>     running egg_info                                                                                               
>     creating /tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info                                                         
>     writing /tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info/PKG-INFO                                                 
>     writing dependency_links to /tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info/dependency_links.txt                 
>     writing top-level names to /tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info/top_level.txt                         
>     writing manifest file '/tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info/SOURCES.txt'                              
>     reading manifest file '/tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info/SOURCES.txt'                              
>     writing manifest file '/tmp/pip-pip-egg-info-3f15zffe/pyads.egg-info/SOURCES.txt'                              
>   Source in /tmp/pip-install-xl4avn05/pyads_79d9cc3840c64db4b21093943835add6 has version 3.3.1, which satisfies requirement pyads==3.3.1 from https://files.pythonhosted.org/packages/58/d7/3194cfc564444685bdcbf94434e3c2ac82424894a9fae8a17ffe4cb9b665/pyads-3.3.1.tar.gz#sha256=e0d010c620e3e5a47c36035d7d43cdf3d1e9cc9a27c99de128485690b7824508
[... cut ...] "'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-i782ir6j                                                                                                       
>   running bdist_wheel                                                                                              
>   running build                                                                                                    
>   **make: *** adslib: No such file or directory.  Stop.**

stlehmann, 2021-01-13:
Thanks for investigating. I think I accidentally did not clone the subrepo adslib on the PC where I made the Push from. I drafted a new release that should now contain the adslib directory. The issue should be fixed by this.

pbruenn, 2021-05-31:
Is it possible this happend again with 3.3.5? I get this message with 3.3.5, were I don't get it with 3.3.4. my host is Linux
Traceback (most recent call last):
  File "/builds/beckhoff/AdsLib/tools/add-route.py", line 3, in <module>
    import pyads
  File "/usr/local/lib/python3.9/dist-packages/pyads/__init__.py", line 10, in <module>
    from .ads import (
  File "/usr/local/lib/python3.9/dist-packages/pyads/ads.py", line 57, in <module>
    from .pyads_ex import (
  File "/usr/local/lib/python3.9/dist-packages/pyads/pyads_ex.py", line 88, in <module>
    _adsDLL = ctypes.CDLL(adslib)
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: adslib.so: cannot open shared object file: No such file or directory

stlehmann, 2021-06-01:
@pbruenn thanks for reporting. Just confirmed.

pbruenn, 2021-06-01:
@stlehmann I don't know anything about python, but maybe GitHub actions can be an option to have defined release process?
I was surprised how easy it was to use them https://github.com/Beckhoff/ADS/blob/master/.github/workflows/build.yml

If you can't store secrets, you need to push to the pip repository, you could just use the actions to build the artifacts, which you then could upload manually to pip. Just an idea.

stlehmann, 2021-06-01:
In fact I just changed CI to use Github Actions and this it what finally caused the problem 😅. Or better say: I caused the problem by not including sumodules in the Github Actions workflow. I just fixed this so it shouldn't be an issue anymore. @pbruenn please try the new version 3.3.7.

pbruenn, 2021-06-01:
Haha, okay sorry for my "smart" advice, I should have looked that up first. At least I can confirm 3.3.7 works fine for me again. Thanks a lot for that fast response!

stlehmann, 2021-06-01:
Great, so I'll close this issue again and hope it stays this way 😊

closed 2021-06-01 by stlehmann

henning-meinhardt, 2021-12-29:
@stlehmann I'm having similar trouble with pyads when trying to load it from AppDaemon. There is an addon that embeds AppDaemon with HomeAssistant. This addon allows to specify additional python libs to be installed and make available to AppDaemon. When specifying the pyads library there, the addon loads it just fine, but unfortunately version 3.2.<some> which does not include the feature of listing all symbols - you added that in 3.3.0. the AddDaemon addon allows to specify a specific version of the pyads to load. But trying to load 3.3.0, 3.3.5, 3.3.7 or 3.3.9 (these are the ones I checked) leads to an error when installing the library (as there is no "wheel" - whatever that is, I'm a Java guy with no much knowledge of python). When trying to import the library in a python script, another error is raised saying that the shared system library adslib is not available. So it seems that installation failed.

AppDaemon in running on a linux system, using python 3.9. I't in a docker container, I guess alpine based (but did not dig deeper).

I filed an issue at https://github.com/hassio-addons/addon-appdaemon/issues/180 showing the error messages in detail.

Can you tell me if the problem is with your library, the AppDaemon addon's mechanism of installing additional libraries or on my site?

Kind regards
Henning

stlehmann, 2021-12-30:
As described in https://github.com/hassio-addons/addon-appdaemon/issues/180#issuecomment-1002901730 I think there is just the wheel package missing in your AppDaemon. Try adding it to the configuration and see if it works.

henning-meinhardt, 2021-12-30:
@stlehmann tried that out and reported in mentioned issue #180. unfortunately still the same problem. I'm wondering why loading an old version 3.2.2 just works fine as
```
Looking in links: https://wheels.home-assistant.io/alpine-3.13/amd64/
Collecting pyads
  Downloading https://wheels.home-assistant.io/alpine-3.13/amd64/pyads-3.2.2-py3-none-any.whl (509 kB)
Installing collected packages: pyads
Successfully installed pyads-3.2.2
```

NaysayerWasTaken, 2025-05-15:
@fcervigni-EP 

I encountered this on python:3.13-alpine today while using pyads v3.4.2.

I managed to workaround this issue by pulling the prebuilt libraries from the [Beckhoff ads releases](https://github.com/Beckhoff/ADS/releases) and placing them in the shared libraries (/usr/local/lib) manually.