stlehmann/pyads #189: AdsSymbol.read() raises NoneType Error for Arrays of Structs

stlehmann, 2021-01-29:
I tried the new AdsSymbol class and ran into an issue when trying to read an element of an array of structs.

I defined the following global variable in TwinCAT 3:

    a: ARRAY[0..1000] of ErrorStruct;

ErrorStruct is defined like this:

```
TYPE ErrorStruct :
STRUCT
	error_time: DT;
	error_text: STRING;
END_STRUCT
END_TYPE
```

I create a symbol for the error_text member of an array element and try to read the value:

```python
>>> import pyads
>>> plc = pyads.Connection("127.0.0.1.1.1", 851)
>>> plc.open()
>>> symbol = plc.get_symbol("GVL.a[0].error_text")
>>> symbol.read()

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-43-07bce4d900ad> in <module>
----> 1 symbol.read()

C:\anaconda3\lib\site-packages\pyads\symbol.py in read(self)
    144         """
    145         self._read_write_check()
--> 146         self.value = self._plc.read(self.index_group, self.index_offset, self.plc_type)
    147         return self.value
    148 

C:\anaconda3\lib\site-packages\pyads\ads.py in read(self, index_group, index_offset, plc_datatype, return_ctypes, check_length)
    689                 plc_datatype,
    690                 return_ctypes,
--> 691                 check_length,
    692             )
    693 

C:\anaconda3\lib\site-packages\pyads\pyads_ex.py in adsSyncReadReqEx2(port, address, index_group, index_offset, data_type, return_ctypes, check_length)
    762         data = (STRING_BUFFER * PLCTYPE_STRING)()
    763     else:
--> 764         data = data_type()
    765 
    766     data_pointer = ctypes.pointer(data)

TypeError: 'NoneType' object is not callable
```

The same goes for accessing the parent variables:

```
>>> symbol = plc.get_symbol("GVL.a[0]")
>>> symbol = plc.get_symbol("GVL.a")
```

I'll do some digging later on but maybe @RobertoRoos already has an idea concerning this issue.

stlehmann, 2021-01-29:
Could this be related to https://github.com/stlehmann/pyads/issues/183?

RobertoRoos, 2021-01-29:
What are the registered `sym_type` and `type_name`? It looks it cannot determine the correct type.
Could you use the debugger to find the result of `SymbolInfoEx`?

And if you specify the type and address manually, does it read/write normally?

stlehmann, 2021-01-29:
The symbol_type is rightly "ErrorStruct" but plc_type is None.

[image]

stlehmann, 2021-01-30:
Using address and type raises the same error:

```python
>>> symbol_by_adr = plc.get_symbol(index_group=16448, index_offset=385000, symbol_type='ARRAY [0..10] OF ErrorStruct')
>>> symbol_by_adr.read()

---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-17-b158a0029ad1> in <module>
----> 1 symbol_by_adr.read()

~\anaconda3\lib\site-packages\pyads\symbol.py in read(self)
    144         """
    145         self._read_write_check()
--> 146         self.value = self._plc.read(self.index_group, self.index_offset, self.plc_type)
    147         return self.value
    148 

~\anaconda3\lib\site-packages\pyads\ads.py in read(self, index_group, index_offset, plc_datatype, return_ctypes, check_length)
    682         """
    683         if self._port is not None:
--> 684             return adsSyncReadReqEx2(
    685                 self._port,
    686                 self._adr,

~\anaconda3\lib\site-packages\pyads\pyads_ex.py in adsSyncReadReqEx2(port, address, index_group, index_offset, data_type, return_ctypes, check_length)
    762         data = (STRING_BUFFER * PLCTYPE_STRING)()
    763     else:
--> 764         data = data_type()
    765 
    766     data_pointer = ctypes.pointer(data)

TypeError: 'NoneType' object is not callable

```

chrisbeardy, 2021-02-01:
I dare say this is related to #183 . Getting the PLC type of a structure does not appear to be possible / or is not provided in the getsymbolinfo layer. I would expect trying to read the whole structure to not work (without needing to modify the code to pass through the sturcture def and then unpacking the bytes). But getting the symbol and reading the variable of an individual variable in the structure, I would instincively would have thought was possible and that the symbolinfo would have found the right type for the symbol info. have you trying making the structure pack mode  to see if it can find the individual variable symbol correctly?

stlehmann, 2021-02-10:
Not yet, I'll give it a go today.

koppa, 2021-02-17:
A potential fix, but I would write some unit tests before doing a pull request:

```
diff --git a/pyads/symbol.py b/pyads/symbol.py
index 8e2be8a..e6e43be 100644
--- a/pyads/symbol.py
+++ b/pyads/symbol.py
@@ -54,7 +54,7 @@ class AdsSymbol:
         name: Optional[str] = None,
         index_group: Optional[int] = None,
         index_offset: Optional[int] = None,
-        symbol_type: Optional[str] = None,
+        symbol_type: Any = None,
         comment: Optional[str] = None,
         auto_update: bool = False,
     ) -> None:
@@ -89,7 +89,7 @@ class AdsSymbol:
             if name is None:
                 raise ValueError(
                     "Please specify either `name`, or `index_group`, "
-                    "`index_offset` and plc_type"
+                    "`index_offset` and symbol_type"
                 )

         self.name = name
@@ -108,7 +108,13 @@ class AdsSymbol:

         self.plc_type: Optional[Any] = None
         if self.symbol_type is not None:
-            self.plc_type = AdsSymbol.get_type_from_str(self.symbol_type)
+            # we get it here as string or integer or as ctypes like type
+            try:
+                # we need to determine whether it is a ctypes type based on the private class _CData
+                sizeof(self.symbol_type)
+                self.plc_type = self.symbol_type # ctypes type
+            except:
+                self.plc_type = AdsSymbol.get_type_from_str(self.symbol_type)

         self.auto_update = auto_update

@@ -130,7 +136,8 @@ class AdsSymbol:
         # However, this type ignores whether the variable is really an array!
         # So are not going to be using this and instead rely on the textual
         # type
-        self.symbol_type = info.symbol_type  # Save the type as string
+        if self.symbol_type is None:
+            self.symbol_type = info.symbol_type  # Save the type as string

     def _read_write_check(self) -> None:
         """Assert the current object is ready to read from/write to.
```

RobertoRoos, 2021-02-17:
I would suggest using some `if-else` instead of `try-catch`:

```
if isinstance(self.symbol_type, str):
    self.plc_type = AdsSymbol.get_type_from_str(self.symbol_type)
elif isinstance(self.symbol_type, _CDate):
    self.plc_type = self.symbol_type  # No conversion needed
else:
    # Raise exception maybe?
```

RobertoRoos, 2021-02-17:
This whole discussions thread is getting out of hand.

Anyway, @stlehmann , getting _a single field_ from a struct (or an array of structs) works fine with `AdsSymbol`.
However, neither the `DT` nor the `STRING(...)` types have been mapped to Python types, hence the the symbol class cannot create a map. I'll see I can whip up a fix right now.

stlehmann, 2021-02-17:
I think at least the first part of the issue is fixed with #206. I would like to have structures fully supported for Symbols in the future as well. I will creat a separate issue for this.

closed 2021-02-17 by stlehmann